FeitianSmartcardReader/FEITIAN_MOBILE_READERS #3: bR301 with Android

AppWerft, 2018-08-02:
1. On the package I see Made for i*, but here I see an Android SDK.
2. I cannot pair with LG6 either Nokia8. (does I need firmwarew update)

Best regards
Rainer

FeitianSmartcardReader, 2018-08-03:
@AppWerft May I know which package show Made for i*, the bR301 reader certified by MFi, so it should have Made for iPhone/iPad/iPod works in SDK.

And may I know which reader are you using? is bR301(Blue color with BT3.0) or bR301BLE(black color with Bluetooth Smart)?

For bR301BLE, only support pair in Application, so please install our APK and do pair in SDK, thanks

FeitianSmartcardReader, 2018-12-05:
@AppWerft Rainer, how is everything going?

closed 2019-01-10 by FeitianSmartcardReader

AppWerft, 2019-01-10:
Hi,
I have the bR301BLE. I try to build an own app. You wrote that I cannot pair in system configuration, I will try in my app. Thanks

AppWerft, 2019-04-12:
Sorry for delay.
Today I have installed the newest apk. On top title I see `FTReaderStandard`. I see:
Hello World  LibVersion 0.7.00.
I select PC/SC API and BT3.0. The list of devices remains empty.  The same behavior I observe in the 3 other cases (Private API and BT and BTLE). 
The blue device with health card is nearby and the LED is flashing. Whats going wrong?