FeitianSmartcardReader/FEITIAN_MOBILE_READERS #4: bR301 with iOS

bantunesptech, 2018-12-03:
I used your sample project (iReader_Basic), with libiRockey301_ccid_3.5.30_Debug and I changed it to send the following commands:

```
[self sendCommand:@"00A4040007604632FF000002"];
[self sendCommand:@"00A4000C022F00"];
[self sendCommand:@"00A4080C045F00EF02"];
[self sendCommand:@"00B0000000"];
```

Everything is fine, until the last command (got this error: "0x80100016"), and got the delegate telling that the card was removed (attached = NO).

But when I use the iR301-U reader everything is just fine.

Here's the output of the commands (bR301):

```
Sending Command: 00A4040007604632FF000002
==========write:
6f0c0000 00000300 000000a4 04000760 4632ff00 0002
==========read:
80020000 00000300 00009000
Rev:<9000>

Sending Command: 00A4000C022F00
==========write:
6f070000 00000400 000000a4 000c022f 00
==========read:
80020000 00000400 00009000
Rev:<9000>

Sending Command: 00A4080C045F00EF02
==========write:
6f090000 00000500 000000a4 080c045f 00ef02
==========read:
80020000 00000500 00009000
Rev:<9000>

Sending Command: 00B0000000
==========write:
6f050000 00000600 000000b0 000000
0x80100016:An attempt was made to end a non-existent transaction.
```

And here's the output of the (iR301-U):

```
Sending Command: 00A4040007604632FF000002
==========write:
6f0c0000 00000900 000000a4 04000760 4632ff00 0002
==========read:
80020000 00000900 00009000
Rev:<9000>

Sending Command: 00A4000C022F00
==========write:
6f070000 00000a00 000000a4 000c022f 00
==========read:
80020000 00000a00 00009000
Rev:<9000>

Sending Command: 00A4080C045F00EF02
==========write:
6f090000 00000b00 000000a4 080c045f 00ef02
==========read:
80020000 00000b00 00009000
Rev:<9000>

Sending Command: 00B0000000
==========write:
6f050000 00000f00 000000b0 000000
==========read:
80020100 00000f00 00005265 ...
Rev:<526570c3 ba626c69 63612050 ...>
```
Any help would be appreciated. Thanks.

FeitianSmartcardReader, 2018-12-04:
@bantunesptech Is possible using new lib 3.5.38? and get a log for us? if possible, can we arrange remote session allow our R&D check? you also can send mail to me hongbin@ftsafe.com, thanks

bantunesptech, 2018-12-04:
@FeitianSmartcardReader unfortunately the error is still there. Same input and same output as before. Thanks.

FeitianSmartcardReader, 2018-12-05:
@bantunesptech as we talked, the bR301 is from 2013, and the firmware we don't maintain anymore, will provide new sample to you for evaluate, thanks

closed 2018-12-05 by FeitianSmartcardReader