FeitianSmartcardReader/FEITIAN_MOBILE_READERS #5: ignoring file libiRockey301_ccid_V3.5.40_debug.a, missing required architecture x86_64

iPopular, 2019-02-21:
Hello, I got error when build project detail as below 

`ld: warning: ignoring file /Users/ipopular/Documents/Project/Freelance/Brainstem/FEITIAN_MOBILE_READERS-master/iOS SDK/iReader for Appstore/Swift Demo/iReader/Sdk/lib/libiRockey301_ccid_V3.5.40_release.a, missing required architecture x86_64 in file /Users/ipopular/Documents/Project/Freelance/Brainstem/FEITIAN_MOBILE_READERS-master/iOS SDK/iReader for Appstore/Swift Demo/iReader/Sdk/lib/libiRockey301_ccid_V3.5.40_release.a (4 slices)ld: warning: ignoring file /Users/ipopular/Documents/Project/Freelance/Brainstem/FEITIAN_MOBILE_READERS-master/iOS SDK/iReader for Appstore/Swift Demo/iReader/Sdk/lib/libiRockey301_ccid_V3.5.40_debug.a, missing required architecture x86_64 in file /Users/ipopular/Documents/Project/Freelance/Brainstem/FEITIAN_MOBILE_READERS-master/iOS SDK/iReader for Appstore/Swift Demo/iReader/Sdk/lib/libiRockey301_ccid_V3.5.40_debug.a (4 slices)

Undefined symbols for architecture x86_64:
  "_SCardEstablishContext", referenced from:
      iReader_swift.ScanDeviceController.viewDidLoad() -> () in ScanDeviceController.o
  "_FtGetSerialNum", referenced from:
      closure #2 () -> () in closure #1 () -> () in iReader_swift.DeviceInfoTableViewController.getDeviceInfo() -> () in DeviceInfoTableViewController.o
  "_FtGetDevVer", referenced from:
      closure #1 () -> () in closure #1 () -> () in iReader_swift.DeviceInfoTableViewController.getDeviceInfo() -> () in DeviceInfoTableViewController.o
  "_FtGetLibVersion", referenced from:
      closure #3 () -> () in closure #1 () -> () in iReader_swift.DeviceInfoTableViewController.getDeviceInfo() -> () in DeviceInfoTableViewController.o
  "_OBJC_CLASS_$_ReaderInterface", referenced from:
      objc-class-ref in ScanDeviceController.o
  "_FtGetDeviceUID", referenced from:
      closure #1 () -> Swift.Int32 in iReader_swift.CardCommandViewController.readUID() -> () in CardCommandViewController.o
  "_SCardListReaders", referenced from:
      closure #1 () -> () in iReader_swift.ScanDeviceController.scanDevice() -> () in ScanDeviceController.o
  "_FtReadFlash", referenced from:
      closure #1 () -> Swift.Int32 in iReader_swift.CardCommandViewController.readFlash() -> () in CardCommandViewController.o
  "_SCardDisconnect", referenced from:
      closure #1 () -> Swift.Int32 in iReader_swift.CardCommandViewController.disConnectCard() -> () in CardCommandViewController.o
  "_FtWriteFlash", referenced from:
      closure #1 () -> Swift.Int32 in iReader_swift.CardCommandViewController.writeFlash() -> () in CardCommandViewController.o
  "_SCardConnect", referenced from:
      closure #1 () -> Swift.Int32 in iReader_swift.CardCommandViewController.connectReader() -> () in CardCommandViewController.o
  "_SCardTransmit", referenced from:
      closure #1 () -> Swift.Int32 in iReader_swift.CardCommandViewController.sendTheCommand(command: Swift.String) -> () in CardCommandViewController.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

Help me please!

Thank for you support.

FeitianSmartcardReader, 2019-02-21:
@iPopular Thanks for your feedback, will add it on tomorrow, will update you then, thanks

iPopular, 2019-02-21:
@FeitianSmartcardReader Thank you so much.

iPopular, 2019-02-26:
@FeitianSmartcardReader Are you Updated SDK for this issue right?

FeitianSmartcardReader, 2019-02-26:
@iPopular Here it is, have a check, thanks 
[libiRockey301_ccid_V3.5.42.zip](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/files/2904609/libiRockey301_ccid_V3.5.42.zip)

FeitianSmartcardReader, 2019-04-14:
Latest 3.5.45 already build x86 and x64 support, close the issue.

closed 2019-04-14 by FeitianSmartcardReader