FeitianSmartcardReader/FEITIAN_MOBILE_READERS #6: The IOS app doesn't detect the BR301

farsiss, 2019-03-13:
I've tested your app demo which was available on appstore to see how it works.
But it doesn't detect the BR301 reader. I precise that the BR301 was switched on.

FeitianSmartcardReader, 2019-03-14:
@farsiss could you please show me your device information? 
check Settings->General->About->scroll down, there has device information under SEID, the OEM version cannot detect by our app.

Thanks and looking forward to your reply,

farsiss, 2019-03-14:
I've tested on the ipad air 1 but it doesn't have a SEID and on a Iphone s7, here is its SEID : 042D16EB103E80016309018038939903E06186FC5C88ED47

FeitianSmartcardReader, 2019-03-15:
After you paired with your iOS device, you can find the device information Settings->General->About->scroll down, can you see the device there?
[image]
[image]

farsiss, 2019-03-15:
[image]

FeitianSmartcardReader, 2019-04-14:
@farsiss Your reader is OEM by Zetes, please contact Zetes to have their support, our standard SDK doesn't support, thanks

closed 2019-04-14 by FeitianSmartcardReader