FeitianSmartcardReader/FEITIAN_MOBILE_READERS #7: Cannot connect with my device

AppWerft, 2019-04-13:
Today I have installed the newest apk. 
On top title I see FTReaderStandard. I see: `Hello World LibVersion 0.7.00.`

I select PC/SC API and BT3.0. The list of devices remains empty. The same behavior I observe in the 3 other cases (Private API and BT and BTLE).
The blue device with health card is nearby and the LED is flashing. Whats going wrong?

Second question: I can use the C++ Lib or the Jar. What is the difference, advantage/disadvantage?

AppWerft, 2019-04-13:
OK, the issue with private API and C++ I have understood.
My problem I cannot connect to reader. 

I guess the demo app has no Location permission and that is the reason. Did forgot?

FeitianSmartcardReader, 2019-04-14:
The C++ LIB is simulated with PC/SC API, which allows a user to build their App based on C/C++ directly without changing their source code.

And for Jar, this is simply API for the developer, easy to deploy.

You can choose one of them to use it And for the permission, I remember it already supported and the APK tested by our engineer, I will take a look tomorrow and back to you then, thanks

AppWerft, 2019-04-14:
Now I have wrote an own app (with location permission) ;-). With this code I can find my reader, but after call of openReader() the app crashes.

My code:
```java
case DK.BT3_NEW:
	BluetoothDevice dev1 = (BluetoothDevice) msg.obj;
	Log.d(LCAT, "Device found: " + dev1.getName());
	String[] readerNames;
	try {
		Log.d(LCAT, "try readerOpen() " );
		readerNames = ftReader.readerOpen(dev1);
		Log.d(LCAT, "readerOpened " +readerNames.length);
```
ca. 5 sec after the open command I see the code below on sonsole:

```java
[DEBUG] :  CCIDScheme: (*init* (ftReaderInf ::FTReaderInf) (handler ::Handler)) #!void
[INFO] :   I/werft.leicadem: Compiler allocated 4MB to compile void kawa.standard.Scheme.initScheme()
[DEBUG] :  D/🧩TiFeitian: (main) [109,9944] [CCIDSchemeLog]:(*init* (ftReaderInf ::FTReaderInf) (handler ::Handler)) #!void
[DEBUG] :  D/🧩TiFeitian: (main) [2,9946] Device found: FT__3481F4048A73
[DEBUG] :  D/🧩TiFeitian: (main) [1,9947] try readerOpen
[WARN] :   BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
[WARN] :   W/System.err: com.ftsafe.bt3.Bt3Exception: read failed, socket might closed or timeout, read ret: -1
[WARN] :   W/System.err: 	at com.ftsafe.bt3.BT3.ft_open(BT3.java:161)
[WARN] :   W/System.err: 	at com.ftsafe.CCIDScheme.readerOpen(CCIDScheme.scm:89)
[WARN] :   W/System.err: 	at com.ftsafe.readerScheme.FTReader.readerOpen(FTReader.java:51)
[WARN] :   W/System.err: 	at de.appwerft.feitian.FeitianModule$1.handleMessage(FeitianModule.java:205)
[WARN] :   W/System.err: 	at android.os.Handler.dispatchMessage(Handler.java:107)
```
My code follows  your demo app. I cannot understand why the connection cannot established. The bonding state is BONDED.

This is my repo. The code realizes a module for Titanium. ;-)

AppWerft, 2019-04-15:
more details:
```java
com.ftsafe.readerScheme.FTReader.readerOpen()
```
is a switch, depending on device type a new CCIDSchema will instantiated, in my case:

```java
new CCIDScheme(new BT3(paramContext, mHandler), mHandler)
```
This BT3 constructor get a new BTdefaultadapter, registers a receiver and starts a thread. As the result I can see the device name `FT__3481F4048A73`.

Now I try to open the device.  `readerOpen` tries `ccidScheme.readerOpen()` and uses the same ccid object as the finder. My question: where is `CCIDScheme.scm` and why timeout?

Best!
Rainer

AppWerft, 2019-04-15:
The error message `read failed, socket might closed or timeout` is not part of private jar, it seems to come from Bluetooth stack:

https://stackoverflow.com/questions/18657427/ioexception-read-failed-socket-might-closed-bluetooth-on-android-4-3

FeitianSmartcardReader, 2019-04-15:
Yes, we know this issue, this is not a library issue, will check with our team with our solution, the android engineer got business trip, should be back this Wednesday, will keep you update, thanks

AppWerft, 2019-04-16:
Thanks for you support. First I see an issue with manifest. If there is no entry for LOCATION_PERMISSION, then I cannot enable in system configuration. In my app I have added this entry. Therefore I "see" the device in handler. But I cannot open. 

This method `ft_open()` in `com.ftsafe.bt3.BT3.java` makes trouble.

```java
public void ft_open(Object paramObject) throws Bt3Exception {
    if (paramObject == null) {
      throw new Bt3Exception("device == null");
    }
    mBlueToothAdapter.cancelDiscovery();
    try {
      if ((mBlueToothSocket != null) && (mBlueToothSocket.isConnected()) && (mBluetoothDevice.getAddress().equals(((BluetoothDevice)paramObject).getAddress()))){
        showLog("donaaaaaaaaaaaaaaaaaaaaaa");
        return;
      }
      mBluetoothDevice = ((BluetoothDevice)paramObject);
      if (mBlueToothSocket != null) {
        mBlueToothSocket.close();
      }
      mBlueToothSocket = mBluetoothDevice.createInsecureRfcommSocketToServiceRecord(UUID.fromString("00001101-0000-1000-8000-00805F9B34FB"));
      mBlueToothSocket.connect();
    }
    catch (Exception localException) {
      throw new Bt3Exception(localException.getMessage());
    }
  }
```

FeitianSmartcardReader, 2019-04-30:
same issue with https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/9

Update the manifest file, problem solved.

closed 2019-04-30 by FeitianSmartcardReader