FeitianSmartcardReader/FEITIAN_MOBILE_READERS #9: reader not detected

snihar2, 2019-04-18:
I downloaded the androidSDK for android studio and had to change 2 things in the build.gradle (module:app) because the ndk and cmake were not supported anymore: here is what I changed ;
cmake {
                cppFlags ""
                arguments '-DANDROID_PLATFORM=android-23',
                        '-DANDROID_TOOLCHAIN=clang', '-DANDROID_STL=c++_static'
            }
            ndk {
                abiFilters 'armeabi-v7a'
            }

I changed also replaced in dependencies "compile files" by 
implementation files('libs/ftsafe_0.7.04.jar')
And then I ran the application. 
Before I did that, I downloaded the demo apk directly and tried it, It worked great. 
Now I would want to make my own app from it. But now that I installed the apk file from the android sdk, the app is running, but when I try to find my reader nothing happens, I have got the following in the logcat : 
04-18 14:20:34.099 12338-12338/com.example.ftReaderStandard D/libFTReaderStandard: SCardEstablishContext ok
04-18 14:20:45.983 12338-12451/com.example.ftReaderStandard D/BluetoothAdapter: startLeScan(): null
04-18 14:20:45.984 12338-12451/com.example.ftReaderStandard E/BluetoothAdapter: LE Scan has already started
04-18 14:20:45.985 12338-12338/com.example.ftReaderStandard D/libFTReaderStandard: SCardListReaders2 error 8010002e
Could you help fix this issue?

snihar2, 2019-04-19:
ok sorry actually it the same issue as "cannot connect to my device" with the permission fro location not granted. What I'm asking myself is why the demo App is working but not android SDK donwloaded from github

FeitianSmartcardReader, 2019-04-19:
@snihar2  Thanks for your feedback.

You are using android studio project or eclipse project?
The APK is made by the last jar version which is 0.7.0, so the source code is different than APK, I will talk to our developer to re-package latest one and upload to GIT.

snihar2, 2019-04-19:
Hi I'm using the android version

snihar2, 2019-04-19:
for info here is the exception thrown
W/Binder: Caught a RuntimeException from the binder stub implementation.
    java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
        at android.os.Parcel.readException(Parcel.java:1620)
        at android.os.Parcel.readException(Parcel.java:1573)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:772)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:324)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
        at android.os.Binder.execTransact(Binder.java:453)

snihar2, 2019-04-25:
When do you think the issue will be solved?

FeitianSmartcardReader, 2019-04-25:
Our engineer take a look, and you are right, after android 6, need to give the location permission, we will modify our demo code, thanks for the feedback,

snihar2, 2019-04-29:
Hi again, 
Do you think the code will be solved in the following days, in the week or later?
I need to know otherwise I will need to put my project on hold up to solving the issue

FeitianSmartcardReader, 2019-04-29:
I will check with our engineer and back to you tomorrow, if they are coming to office, should can be done tomorrow, keep in touch, thanks

FeitianSmartcardReader, 2019-04-30:
@snihar2 Update the AndroidManifest.xml file and add below permission, now it should works.
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

Thanks for your feedback. 

On other hand, our R&D team is working on improve the current SDK and it will come early mid of June.

closed 2019-04-30 by FeitianSmartcardReader

AppWerft, 2019-05-02:
In your demo apk the permission is not part of manifest. therefore the demo apk cannot work.

snihar2, 2019-05-02:
> In your demo apk the permission is not part of manifest. therefore the demo apk cannot work.

Last time I tested, the demo app was working properly and the connection could be made. However the android sdk was not properly configured and the connection couldn't be made because of the lack of permission. I didn't got the time to test the new update of the sdk, but of what I have seen, they just implemented the permission. I still have to change some features in the build.graddle of the app because they are using old libraries or other that are not used anymore, for example they still use compile while it has been replaced with 'implementation' and 'api'.

AppWerft, 2019-05-02:
I downloaded  this APK https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/Android%20SDK/BIN

In system configuration/apps/Feitian/permissions I don't see any permissions.  BT and/or BLE cannot work without location permission. Whats going wrong?

snihar2, 2019-05-02:
Download demo application APK file for Android:
https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/raw/master/Android%20SDK/BIN/FTReaderStandard.apk

so you downloaded from here? I think are no permission include but when I tested it, it worked, I guess
 since they upgraded the librarie it doesn't work anymore

AppWerft, 2019-05-02:
Here: https://developer.android.com/guide/topics/connectivity/bluetooth
Bluetooth permissions

In order to use Bluetooth features in your application, you must declare two permissions. The first of these is BLUETOOTH. You need this permission to perform any Bluetooth communication, such as requesting a connection, accepting a connection, and transferring data.

The other permission that you must declare is either ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION. A location permission is required because Bluetooth scans can be used to gather information about the location of the user. This information may come from the user's own devices, as well as Bluetooth beacons in use at locations such as shops and transit facilities.

snihar2, 2019-05-02:
I believe and I issued the same issue, but I'm just telling you that the demo apk did work for me but when I tested it from the android sdk code and tried to run it on my device from android studio, there it didn't work

snihar2, 2019-05-03:
I get these errors everytime I download the project
14:49	Gradle sync failed: No toolchains found in the NDK toolchains folder for ABI with prefix: mips64el-linux-android (23 s 61 ms)
ERROR: ABIs [armeabi] are not supported for platform. Supported ABIs are [arm64-v8a, armeabi-v7a, x86, x86_64].
Affected Modules: app

ERROR: ABIs [armeabi] are not supported for platform. Supported ABIs are [arm64-v8a, armeabi-v7a, x86, x86_64].
Affected Modules: app

WARNING: Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'.
It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html
Affected Modules: app

snihar2, 2019-05-03:
05-03 14:57:40.809 221-8402/? E/AudioHardwareTiny: getInputRouteFromDevice:device:80000004
05-03 14:57:43.158 221-8402/? E/AudioHardwareTiny: getInputRouteFromDevice:device:80000004
05-03 14:57:44.436 565-632/system_process E/DisplaydConnector: Communications error: java.io.IOException: No such file or directory
05-03 14:57:44.437 565-632/system_process E/DisplaydConnector: Error in NativeDaemonConnector: java.io.IOException: No such file or directory

and when I finally try to simulate it on my tablet, it give these errors