FeitianSmartcardReader/FEITIAN_MOBILE_READERS #10: Cannot detect bR301BLE after integrating the new lib ftsafe_0.7.04.jar

KhalilKhalilKhalil, 2019-05-17:
Hello,

We have our own application based on your SDK.
We have used the version 0.7.00 of the jar and everything works fine.
But now we try to use the last version of the jar (0.7.04), but we have a problem : 
When we call readerFind to scan for BLE devices, and then we turn on our BLE reader, the callback DK.BT4_NEW of the handler is never called.
This problem occurs only on android 6 (we tested under another device with android 4 and there is no problem).
Note that we actually add Location permission to AndroidManifest and then requested permission during Runtime, so obviously it is not related to permissions.
We really appreciate any quick feedback from you, thanks !

Best Regards,

FeitianSmartcardReader, 2019-05-17:
@KhalilKhalilKhalil Do you have log for us? How about readerFind API, does it list the reader with Bluetooth ID? or it found the BLE reader, and cannot connect?

Thanks and looking forward to your reply,

KhalilKhalilKhalil, 2019-05-17:
No it is not an error of connection. It cannot even find the ble reader.
I call readerFind and then the scan starts, and when I turn on the ble reader I receive the callback DK.BT4_NEW which means a new device is found. This acually works with version 0.7.00, but now that same logic don't work with the new version 0.7.04 !!
Here the logs after calling readerFind (as you can see nothing happens after I turn on ble reader).

thanks,

```
 09:07:34.326 27971-27971/com.lexpersona.odisia.android D/SettingsInterface:  from settings cache , name = sound_effects_enabled , value = 1
05-17 09:07:34.343 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: isEnabled
05-17 09:07:34.345 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: isEnabled
05-17 09:07:34.347 27971-27971/com.lexpersona.odisia.android D/Odisia: Device running on api level >= 23, so request access to location (mandatory for Bluetooth Le scan).
05-17 09:07:34.382 27971-27991/com.lexpersona.odisia.android D/Surface: Surface::disconnect(this=0xa8a1fd00,api=1)
05-17 09:07:34.385 27971-27991/com.lexpersona.odisia.android D/GraphicBuffer: unregister, handle(0x9d31ae00) (w:812 h:764 s:816 f:0x1 u:0x000b00)
05-17 09:07:34.387 27971-27991/com.lexpersona.odisia.android I/[MALI][Gralloc]: [-]r_hnd(0x9d31ae00), client(31), share_fd(52)
05-17 09:07:34.387 27971-27991/com.lexpersona.odisia.android D/GraphicBuffer: unregister, handle(0x9d31af40) (w:812 h:764 s:816 f:0x1 u:0x000b00)
05-17 09:07:34.388 27971-27991/com.lexpersona.odisia.android I/[MALI][Gralloc]: [-]r_hnd(0x9d31af40), client(31), share_fd(59)
05-17 09:07:34.388 27971-27991/com.lexpersona.odisia.android D/GraphicBuffer: unregister, handle(0x9d31a860) (w:812 h:764 s:816 f:0x1 u:0x000b00)
05-17 09:07:34.388 27971-27991/com.lexpersona.odisia.android I/[MALI][Gralloc]: [-]r_hnd(0x9d31a860), client(31), share_fd(47)
05-17 09:07:34.389 27971-27991/com.lexpersona.odisia.android E/Surface: getSlotFromBufferLocked: unknown buffer: 0x9d31acc0
05-17 09:07:34.389 27971-27991/com.lexpersona.odisia.android D/mali_winsys: [MALI] win=0xa8a1fd08, native_buffer=0x9d31ac28, fd=-1
05-17 09:07:34.389 27971-27991/com.lexpersona.odisia.android D/GraphicBuffer: unregister, handle(0x9d31acc0) (w:812 h:764 s:816 f:0x1 u:0x000b00)
05-17 09:07:34.389 27971-27991/com.lexpersona.odisia.android I/[MALI][Gralloc]: [-]r_hnd(0x9d31acc0), client(31), share_fd(51)
05-17 09:07:34.389 27971-27991/com.lexpersona.odisia.android D/Surface: Surface::disconnect(this=0xa8a1fd00,api=1)
05-17 09:07:34.395 27971-27971/com.lexpersona.odisia.android D/WindowClient: Remove from mViews: com.android.internal.policy.PhoneWindow$DecorView{8c12650 V.E...... R......D 0,0-684,636}, this = android.view.WindowManagerGlobal@823f61f
05-17 09:07:34.399 27971-27971/com.lexpersona.odisia.android D/ActivityThread: ACT-AM_ON_PAUSE_CALLED ActivityRecord{561e3cb token=android.os.BinderProxy@880a2c1 {com.lexpersona.odisia.android/com.lexpersona.odisia.android.activity.MainActivity}}
05-17 09:07:34.408 27971-27971/com.lexpersona.odisia.android D/ActivityThread: ACT-PAUSE_ACTIVITY handled : 1 / android.os.BinderProxy@880a2c1
05-17 09:07:34.738 27971-27971/com.lexpersona.odisia.android V/ActivityThread: Finishing stop of ActivityRecord{561e3cb token=android.os.BinderProxy@880a2c1 {com.lexpersona.odisia.android/com.lexpersona.odisia.android.activity.MainActivity}}: show=true win=com.android.internal.policy.PhoneWindow@e809894
05-17 09:07:34.738 27971-27971/com.lexpersona.odisia.android D/ActivityThread: ACT-STOP_ACTIVITY_SHOW handled : 0 / android.os.BinderProxy@880a2c1
05-17 09:07:38.203 27971-27971/com.lexpersona.odisia.android D/Odisia: Location permission granted. Yay!
05-17 09:07:38.245 27971-27971/com.lexpersona.odisia.android D/WindowClient: Add to mViews: com.android.internal.policy.PhoneWindow$DecorView{35ac428 V.E...... R.....I. 0,0-0,0}, this = android.view.WindowManagerGlobal@823f61f
05-17 09:07:38.247 27971-27991/com.lexpersona.odisia.android D/OpenGLRenderer: CanvasContext() 0xa89c5800
05-17 09:07:38.249 27971-27971/com.lexpersona.odisia.android D/ViewRootImpl: hardware acceleration is enabled, this = ViewRoot{fa63a7a Scan,ident = 3}
05-17 09:07:38.260 27971-27971/com.lexpersona.odisia.android D/Odisia: Init bluetooth reader Feitian BLE. Check whether the smart card reader has been connected.
05-17 09:07:38.481 27971-27971/com.lexpersona.odisia.android D/CCIDScheme: (*init* (ftReaderInf ::FTReaderInf) (handler ::Handler)) #!void
05-17 09:07:38.481 27971-27971/com.lexpersona.odisia.android D/Odisia: Call readerFind()
05-17 09:07:38.484 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: startLeScan(): null
    getLeState() returning 12
05-17 09:07:38.492 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: getLeState() returning 12
    STATE_ON
05-17 09:07:38.493 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: getLeState() returning 12
05-17 09:07:38.500 27971-27983/com.lexpersona.odisia.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
05-17 09:07:38.503 27971-27971/com.lexpersona.odisia.android I/BluetoothLeScanner: startRegisteration: mLeScanClients={android.bluetooth.BluetoothAdapter$2@9e4dbc9=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@7df02ce}
05-17 09:07:38.503 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: isDiscovering
05-17 09:07:38.504 27971-27971/com.lexpersona.odisia.android D/BluetoothAdapter: 102966767: getState(). Returning 12
[... cut ...]
05-17 09:07:38.604 27971-27991/com.lexpersona.odisia.android I/[MALI][Gralloc]: [+]r_hnd(0x9d31ad60), client(31), share_fd(59)
05-17 09:07:38.604 27971-27991/com.lexpersona.odisia.android D/GraphicBuffer: register, handle(0x9d31ad60) (w:812 h:562 s:816 f:0x1 u:0x000b00)
05-17 09:07:38.615 27971-27991/com.lexpersona.odisia.android I/[MALI][Gralloc]: [+]r_hnd(0x9d31b120), client(31), share_fd(60)
05-17 09:07:38.615 27971-27991/com.lexpersona.odisia.android D/GraphicBuffer: register, handle(0x9d31b120) (w:812 h:562 s:816 f:0x1 u:0x000b00)
```

KhalilKhalilKhalil, 2019-05-20:
Hi,
Could you reproduce this issue, or do you have an idea what the problem is after my explication and logs above?
looking forward to your reply...thanks!

FeitianSmartcardReader, 2019-06-03:
Problem solved, which is Mobile device issue, not the library issue, thanks

closed 2019-06-03 by FeitianSmartcardReader