FeitianSmartcardReader/FEITIAN_MOBILE_READERS #14: Xfr with aspected result runs in an issue

AppWerft, 2020-03-13:
Hello,
all APDU which returns 9000 works as aspected.

This APDU for getting data from card runs in an issue: 

```java
byte[] READ_PD = new byte[] { 0x00, (byte) 0xb0, (byte) (0x80 + 0x01), 0x00, 0x00, 0x00, 0x00 };
ftReader.readerXfr(0, READ_PD);		
```
returns:
```
[com.ftsafe.readerScheme.FTReader:readerXfr]
[apdu send recv error]
[java.lang.Exception: apdu send recv error]
```
Any ideas whats going wrong?

FeitianSmartcardReader, 2020-03-13:
I will feedback to our engineer next Monday, they will back to you then, thanks

AppWerft, 2020-03-14:
here my log:
SEND: 00a4040c07d2760001448000  //  SELECT_MF
RCV: 9000 // OK
SEND: 00a4040c06d27600000102 // SELECT_HCA
RCV: 9000 // OK
SEND: 00b0810002 // SELECT_FILE_PD
RCV: 016c9000  //  ?
SEND: 00b08100000000 // READ_PD
[ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][apdu send recv error][java.lang.Exception: apdu send recv error]

closed 2022-03-10 by FeitianSmartcardReader