FeitianSmartcardReader/FEITIAN_MOBILE_READERS #17: SCardListReaders returns SCARD_E_NO_READERS_AVAILABLE on iOS

dfyx, 2020-09-14:
I've recently upgraded the SDK from version 1.x to 3.5.59 in my iOS project for better compatibility with the latest SDK and since then I've had problems connecting to card readers.

I request bluetooth authorization through `CBCentralManager`. As soon as the state is `poweredOn` and I have authorization, I call `SCardEstablishContext` and `SCardListReaders`. This has worked great in the past but now I get a return value of `SCARD_E_NO_READERS_AVAILABLE` even though `EAAccessoryManager` tells me my bR301 is connected. I can temporarily fix this by disconnecting and reconnecting the card reader but the problem reappears after a while.

I've also seen similar behavior with the iR301 though debugging that is a bit harder since I obviously can't connect my debugger through USB at the same time as the iR301 and wifi debugging isn't possible because our test devices aren't allowed to be in our company network.

I assume with the new SDK I have to explicitly connect to the devices to make them show up in `SCardListReaders` but I haven't quite figured out how to do that and the documentation isn't quite clear on how to do that.

FeitianSmartcardReader, 2020-09-15:
Thanks for your feedback, our engineer working to have check this issue, we will get back to you very soon, keep in touch,

FeitianSmartcardReader, 2020-09-16:
@dfyx did you set reader delegate? in the latest lib, it supports Bluetooth 3.0, Bluetooth Smart and lightning based readers. so in your application, please make sure you did below work:
1. Add corebluetooth framework for Bluetooth smart device, and add ExtrnalAccessory framework for lightning and Bluetooth 3.0 readers.
2. Set the reader delegate, the API can find in ReaderInterface.h
3. Call setDeviceType API to set the reader type, if your reader is bR301 with Blue casing, use IR301_AND_BR301
`typedef enum FTDEVICETYPE{
    EMPTY_DEVICE = 0,      // The reader will scan in background automatically, until find the device and then use this type by default
    IR301_AND_BR301 = 1,    // for lightning and Bluetooth 3.0 readers
    BR301BLE_AND_BR500 = 2, //For Bluetooth Smart readers
}FTDEVICETYPE;`
4. Call SCardEstablishContext and SCardListReaders to list your reader and connect it

We provide demo source code, can find in SDK:
https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/iOS_SDK/demo

There have a version already in Appstore, you can download and have a test too.
https://itunes.apple.com/us/app/smartcard-reader/id525954151?mt=8

smartcard-reader source code: https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/iOS_SDK/demo/iReader_Appstore

Please take a test, any questions, contact me anytime, thanks

FeitianSmartcardReader, 2020-09-21:
@dfyx do you still have issue on this topic? thanks and looking forward your reply,

dfyx, 2020-09-21:
Sorry, can't test it right now as I'm in home office for the next few weeks because of Corona. I'll get back to you as soon as I'm at the office again. For now we resolved the problem by downgrading back to 1.32.3 and it appears to work.

FeitianSmartcardReader, 2020-09-21:
Noted, will keep this open status until have your test result, thanks and take care,

dfyx, 2020-12-28:
Better late than never:

It was a bit tricky to get it working with C# but now I can at least get it to read my card most of the time. Now I occasionally get `SCARD_E_READER_UNSUPPORTED` when calling `SCardConnect` which to my knowledge has never happened before. Feels strange because that should only depend on the reader, not on how I'm talking to it.

FeitianSmartcardReader, 2020-12-28:
@dfyx with bR301BLE and bR500(Bluetooth smart), the reader name is FT_XXXXXXXXXX(which is the name of Bluetooth reader), did you input the right name?

dfyx, 2020-12-28:
I did and as said, it only happens occasionally. In my last test it happened when I read the same card with the same reader for the fourth time in a row. So maybe it has something to do with not freeing some handles correctly, I'm not sure.

dfyx, 2020-12-28:
Alright, my last problem seems to have been a race condition in my app that caused it to not always wait for the callback. It's fixed now. Thanks a lot for your support.

closed 2020-12-28 by dfyx