FeitianSmartcardReader/FEITIAN_MOBILE_READERS #18: Connecting to PIV applet on br301

tperfitt, 2020-10-17:
I am trying to send APDU commands to a PIV card on iOS and i keep getting back the same error:

0x80100016: An attempt was made to end a non-existent transaction

I have tried different cards, all the ios demo apps, and the app in the iOS app store. The reader connects fine over bluetooth and shows the ATR value. I then send the select applet command APDU:

00a404000ba0000003974349445f0100

but the error comes back the same. Everything works fine if I connect the reader directly to my Mac over USB. Is there a way to get direct access to the Bluetooth characteristics or does everything have to go through the SDK? 

The version of iOS is iOS14. I have also tried other cards (credit cards--mainly mastercard) just to see if I could get any APDU commands to work. Only 1 of my credit cards returned anything beside the "non-existent transaction" error.

Please provide any guidance on how to send APDU commands over bluetooth to a PIV card.

FeitianSmartcardReader, 2020-10-19:
@tperfitt did you try our latest release app on Appstore?
https://itunes.apple.com/us/app/smartcard-reader/id525954151?mt=8

Please use this App to do operation, and in the main view, it has log button, use that button can export the log, we will take a look the log and back to you.

The reader based on CCID protocol, if you don't want use our SDK, then you may need do envelope the APDU by yourself, the reader firmware based on CCID standard, which you can find from below:
https://usb.org/sites/default/files/DWG_Smart-Card_CCID_Rev110.pdf

The below 3 commands can complete the APDU transfer, Power ON/OFF and XfrBlock.

[image]

Any questions, contact us anytime, thanks

FeitianSmartcardReader, 2020-10-19:
With Bluetooth UUID:

We released two products, and UUIDs are below:
#define UUIDSTR_bR301C_PROPRIETARY_SERVICE  @"46540001-0002-00C4-0000-465453414645"
#define UUIDSTR_bR301C_TRANS_RX             @"46540002-0002-00C4-0000-465453414645"
#define UUIDSTR_bR301C_TRANS_TX             @"46540003-0002-00C4-0000-465453414645"

#define UUIDSTR_bR301C_PROPRIETARY_SERVICE  @"46540001-0002-00C6-0000-465453414645"
#define UUIDSTR_bR301C_TRANS_RX             @"46540002-0002-00C6-0000-465453414645"
#define UUIDSTR_bR301C_TRANS_TX             @"46540003-0002-00C6-0000-465453414645"

tperfitt, 2020-10-20:
Thank you. I'll export the log, though all it says is "card inserted" then:

0x80100016: An attempt was made to end a non-existent transaction

I'll get the exact log later today and post.

I would like to send APDU command directly the reader without the SDK. Thank you for the UUIDs for the device. Do I just write the APDU commands to the TX and monitor the RX for notifications? Any sample code and information on setting up the connection in preparation for sending APDU commands would be helpful.

tperfitt, 2020-10-21:
Here is the log:

2020-10-20 15:55:21.277 iReader[11068:2974228] IAPDHasLaunched: kIAPAvailableNotification iapdAvailableState 0 -> 0
2020-10-20 15:55:21.278 iReader[11068:2974228] IAP2DHasLaunched: kIAP2AvailableNotification iap2dAvailableState 0 -> 0
2020-10-20 15:55:21.278 iReader[11068:2974228] -[EAAccessoryManager _initFromSingletonCreationMethod] isRunningOnMac
2020-10-20 15:55:21.289 iReader[11068:2974228] On: 956 RegisteredForLocalNotifications Count: 1
2020-10-20 15:55:21.290 iReader[11068:2974254] -----readerStatusThread
2020-10-20 15:55:21.290 iReader[11068:2974247] CBPeripheralManagerStatePoweredOn,sdkversion----3.5.61
2020-10-20 15:55:21.561 iReader[11068:2974228] 0--------->>>>>enter foreground,sdkversion----3.5.61
2020-10-20 15:55:24.213 iReader[11068:2974228] -------------->>>>enter background,sdkversion----3.5.61
2020-10-20 15:55:25.862 iReader[11068:2974228] 0--------->>>>>enter foreground,sdkversion----3.5.61
2020-10-20 15:55:58.848 iReader[11068:2974228] -------------->>>>enter background,sdkversion----3.5.61
2020-10-20 16:00:36.901 iReader[11068:2974228] 0--------->>>>>enter foreground,sdkversion----3.5.61
2020-10-20 16:00:38.013 iReader[11068:2974228] -------------->>>>enter background,sdkversion----3.5.61
2020-10-20 16:00:40.306 iReader[11068:2974228] -------------->>>>enter background,sdkversion----3.5.61
2020-10-20 19:58:04.101 iReader[11068:2974228] 0--------->>>>>enter foreground,sdkversion----3.5.61
2020-10-20 19:58:06.848 iReader[11068:2974660] 
Peripheral Info:NAME: FT_00A050150E30
UUID: 8C7CB08A-2454-40AC-1A34-EADDF02B406C
RSSI: -52
didDiscoverPeripheral
,sdkversion----3.5.61
2020-10-20 19:58:06.854 iReader[11068:3083917] peripheralList1:(
    "<CBPeripheral: 0x281094320, identifier = 8C7CB08A-2454-40AC-1A34-EADDF02B406C, name = FT_00A050150E30, state = disconnected>"
),sdkversion----3.5.61
2020-10-20 19:58:06.854 iReader[11068:3083917] peripheralList:(
    "<CBPeripheral: 0x281094320, identifier = 8C7CB08A-2454-40AC-1A34-EADDF02B406C, name = FT_00A050150E30, state = disconnected>"
),sdkversion----3.5.61
2020-10-20 19:58:07.065 iReader[11068:2974660] centralManager:didConnectPeripheral:,sdkversion----3.5.61
2020-10-20 19:58:07.539 iReader[11068:3083915] peripheral:didDiscoverCharacteristicsForService:error:,sdkversion----3.5.61
2020-10-20 19:58:07.600 iReader[11068:3083915] peripheral:didDiscoverCharacteristicsForService:error:,sdkversion----3.5.61
2020-10-20 19:58:07.600 iReader[11068:3083915] Wirte Channel is open successed,sdkversion----3.5.61
2020-10-20 19:58:07.600 iReader[11068:3083915] Read Channel is open successed,sdkversion----3.5.61
2020-10-20 19:58:07.928 iReader[11068:3083915] rev:{length = 2, bytes = 0x5003},sdkversion----3.5.61
2020-10-20 19:58:07.930 iReader[11068:3083991] write:{length = 14, bytes = 0x6b040000000000000000a55a3730}
2020-10-20 19:58:07.989 iReader[11068:3083915] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:07.989 iReader[11068:3083915] rev:{length = 15, bytes = 0x83050000000000010000040f019000},sdkversion----3.5.61
2020-10-20 19:58:07.989 iReader[11068:3083917] ###########271,sdkversion----3.5.61
2020-10-20 19:58:07.989 iReader[11068:3083917] card present
2020-10-20 19:58:08.017 iReader[11068:3083993] write:{length = 14, bytes = 0x6b040000000001000000a55a3816}
2020-10-20 19:58:08.168 iReader[11068:3083917] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:08.169 iReader[11068:3083917] rev:{length = 20, bytes = 0x830e00000000010100000c036200520033003000},sdkversion----3.5.61
2020-10-20 19:58:08.169 iReader[11068:3083917] rev:{length = 4, bytes = 0x31009000},sdkversion----3.5.61
2020-10-20 19:58:08.169 iReader[11068:3083995] write:{length = 10, bytes = 0x65000000000002000000}
2020-10-20 19:58:08.227 iReader[11068:3083917] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:08.229 iReader[11068:3083917] rev:{length = 10, bytes = 0x81000000000002010000},sdkversion----3.5.61
2020-10-20 19:58:08.230 iReader[11068:3083997] write:{length = 10, bytes = 0x62000000000003000000}
2020-10-20 19:58:08.288 iReader[11068:2974249] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:08.408 iReader[11068:3083917] rev:{length = 20, bytes = 0x801600000000030000003bfc1800008131804590},sdkversion----3.5.61
2020-10-20 19:58:08.408 iReader[11068:3083917] rev:{length = 12, bytes = 0x67464a00641606f2727e00e0},sdkversion----3.5.61
2020-10-20 19:58:11.417 iReader[11068:3084011] write:{length = 14, bytes = 0x6b040000000004000000a55a3400}
2020-10-20 19:58:11.468 iReader[11068:3083917] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:11.468 iReader[11068:3083917] rev:{length = 14, bytes = 0x8304000000000400000012019000},sdkversion----3.5.61
2020-10-20 19:58:11.468 iReader[11068:3084013] write:{length = 14, bytes = 0x6b040000000005000000a55a3813}
2020-10-20 19:58:11.527 iReader[11068:2974249] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:11.527 iReader[11068:3083917] rev:{length = 20, bytes = 0x8312000000000500000010034600650069007400},sdkversion----3.5.61
2020-10-20 19:58:11.529 iReader[11068:2974249] rev:{length = 8, bytes = 0x690061006e009000},sdkversion----3.5.61
2020-10-20 19:58:11.529 iReader[11068:3084015] write:{length = 14, bytes = 0x6b040000000006000000a55a3231}
2020-10-20 19:58:11.587 iReader[11068:3083917] Send Successfully,sdkversion----3.5.61
2020-10-20 19:58:11.587 iReader[11068:2974249] rev:{length = 20, bytes = 0x830a00000000060000003cf6a2f6991a00019000},sdkversion----3.5.61
[... cut ...]
AF 94 FB ED 67 DA 3B 17 F4 4F B7 4C EE EC 8B CC 
93 35 5F 95 E2 A5 B6 FB F1 AD 7D 2A 93 8F D7 58 
38 F9 CF 5E 79 C9 B8 FF FF 3E 75 F1 E3 51 8F 84 
8B F9 CB 4F 9F 78 9A 92 B2 62 85 8D 00 AB BE CC 
05 B5 B8 E5 F2 AE DE E7 F7 64 5C 4A 3D 1B BB 57 
86 57 E0 E6 99 6C 3E 5E ED 5B F7 A6 C7 F2 45 C8 
61 2B : 0x2B bytes of response still available.
00 C0 00 00 2b
> 00 C0 00 00 2b
< 6F 97 97 BC B8 85 C5 62 61 52 E7 C6 99 7C 0B BA 
E6 AC 38 5F F9 69 CA F1 C0 EF 2C 7E 6F 00 95 5F 
A2 B3 96 04 00 00 71 01 01 FE 00 90 00 : Normal processing.

FeitianSmartcardReader, 2020-10-21:
Will let our engineer to write a simple demo code tell you how to do it.

In my viewpoint, you will need handle below steps by yourself:
1. Search UUID and pair with the bR301BLE
2. Add notification to monitoring the data input from reader, in passive mode, the reader will return the card slot status when trigger the switch in card slot.
*5002 means the card absent, 5003 means the card inserted
other data, take a look the CCID specification of response part, the head of CCID format is 0x80
3. If detect the card inserted, then follow the CCID specification send the 62(Card power ON)/6F(Data communicate)/63(Card Power OFF) commands to communicate with card. the format is 10 bytes CCID head + APDU
If you  to check the card slot status in active mode, then send the 65 command to get the slot status.

tperfitt, 2020-10-21:
After looking at the log and the spec, I was able to read and write to the bluetooth characteristics and get responses. I get notified of insert and removal events. So I now understand how to get info and write apdu commands. The select works but I am still getting the same response. 

If I write :

00 cb 3f ff 05 5c 03 5f c1 01

 with a physical reader, I get back:

53 82 04 27 70 82 04 1E 1F 8B 08 00 00 00 00 00
00 0B 33 68 62 99 64 D0 C4 5C B5 80 99 89 91 89
89 4B 2C E2 65 27 03 23 03 83 61 A5 01 2F 1B A7
56 9B 47 DB 77 5E 46 46 6E 56 06 83 68 43 61 03
41 36 2E CE 49 6A 9D 93 3F E9 A4 30 4A 8A 31 27
E7 E7 1A 8A 19 88 A0 08 B2 15 64 96 65 A7 56 1A
EA 18 68 B1 31 87 B2 30 0B 2B 07 78 86 79 A7 56
2A B8 A4 96 65 26 A7 2A 38 A7 16 95 64 A6 65 26
27 96 A4 2A 38 96 96 64 E4 17 65 96 54 1A C8 89
F3 1A 5A 18 98 18 99 19 19 99 98 9B 98 45 89 F3
1A 1B 00 B9 C6 50 AE 81 91 A1 81 81 1E C4 3C 75
A8 79 8E 46 A6 16 8E 4E 8E 8E 96 46 16 8E AE 06
26 2E 16 86 2E 16 06 06 6E A6 A6 2E 06 E6 46 6E
66 C6 06 4D 8C 4A C8 7E 60 64 65 60 6E 62 E4 67
00 8A 73 31 35 31 32 32 EC D2 68 FB 78 D8 EE A4
6A DF 9F 67 37 73 8C BA D7 BE D2 5B DD F7 47 73
61 00 : 0x00 bytes of response still available.

If I encode the package and send to the Feitian bluetooth reader:
0x8000000000000f40fe00

I get back:

0x8000000000001040fe00

This is the same response I got with the app. The select command works fine and I get back the same response so I suspect I am encoding it correctly.  Is there any setup commands and/or modes that it need to be put into?

FeitianSmartcardReader, 2020-10-22:
It should be like this:
`62000000000006000000
6f10000000000700000000a404000ba000000308000010000100
6f0a000000000800000000cb3fff055c035fc10b`

And if you got 61XX, then send `00C00000XX` to get response data

FeitianSmartcardReader, 2020-10-22:
And BTW, in your code, you don't need to care the reader protocol, with Bluetooth part, our firmware will handle the T0 and T1 protocol, you just need encode your APDU as CCID format.
> CCID head + APDU
> 6f100000000007000000+00a404000ba000000308000010000100

FeitianSmartcardReader, 2020-10-22:
The response data -  explain in CCID spec, chapter 6.2.1:
0x80000000000010 40 fe 00

0x40: check the CCID spec, chapter 6.2-3, which means the card detected and inactive
0xFE: chapter 6.2-3 CCID timed out while talking to the ICC

tperfitt, 2020-10-22:
I just decoded that as well. The Power On gives the correct ATR response, but any other messages give the 0x40FE:
Send: 62
00000000
00
06
00
0000

Received: 0x831111AC83009000 (ATR)
--------

PC_to_RDR_XfrBlock
Sent: 
bMessageType:6f
dwLength:10000000 
bSlot: 00
bSeq: 07
bBWI: 00
wLevelParameter: 0000
Data: 00a404000ba000000308000010000100 (16 bytes apdu for select)

Received: 
bMessageType: 0x80
dwLength: 00000000
bSlot:00
bSeq:07
bStatus:40
bError: FE (ICC_MUTE -- CCID timed out while talking to the ICC)
bChainParameter: 00
abData:

FeitianSmartcardReader, 2020-10-22:
Sorry, I forgot inform you, if the data length more than 20 bytes, you need split them, the BLE only can accept 20bytes each time

With below command, send two times:
6f10000000000700000000a404000ba000000308000010000100
First time: 6f10000000000700000000a404000ba000000308
Second time:000010000100

tperfitt, 2020-10-22:
The timeout is between the reader and the smartcard, right? Doesn't that imply a timing issue that I don't have control over?

tperfitt, 2020-10-22:
I found this on: https://ccid.apdu.fr/ccid/unsupported.html#0x09C30x0008

Seems very similar:
USB descriptor: readers/Feitian_SCR301.txt

The reader fails with a CASE 2 APDU with Le=0 (256 bytes)
Sent: 80 34 01 00 00
ifdhandler.c:1219:IFDHTransmitToICC() no name (lun: 0)
commands.c:1542:CmdXfrBlockTPDU_T0() T=0: 5 bytes
-> 000000 6F 05 00 00 00 00 07 00 00 00 80 34 01 00 00
<- 000000 80 00 00 00 00 00 07 40 FE 00
commands.c:1316:CCID_Receive Card absent or mute
And time extension requests are not forwarded back to the host.
A new version of the reader is available and may fix this issue.

FeitianSmartcardReader, 2020-10-22:
In my viewpoint, you haven't send the complete data to the reader, the BLE reader only accept first 20 bytes data, so if your data more than 20bytes, then you need to split multi-packet.

Could you please try split to multi-packet and have a test? And I am working to write demo code on my PC, back to you later.

tperfitt, 2020-10-22:
I didn't see you comment about splitting up the packet before I posted my last post. Sorry about that. I tried splitting the packet up into 2 packages like you suggested and I didn't get any response. I send the CmdPowerOn, get a response, then send the two packets and I am was not notified of any data returned.

FeitianSmartcardReader, 2020-10-22:
can please send mail to me? I share some code to you in mail
send to hongbin@ftsafe.com

thanks

tperfitt, 2020-10-22:
Email sent!

tperfitt, 2020-10-22:
I received the email and before I tried the code, I thought I would pair the reader with a Windows 10 PC and make sure that the reader could see the card and read the certificates on the card. It could not and appears to have the same error. It can read the ATR but cannot get pass that. I then plugged in a USB smart card reader and inserted the same card into that reader and ran the same command and it was able to read it fine. Below is the output:

TEST 1: Installed driver, paired to Bluetooth reader BR301 and ran certutil -scinfo
C:\Users\tperf>certutil -scinfo
The Microsoft Smart Card Resource Manager is running.
Current reader/card status:
Readers: 1
    0: Avtor KP375-BLE 0
--- Reader: Avtor KP375-BLE 0
--- Status: SCARD_STATE_EMPTY
--- Status: No card.
---   Card:

=======================================================
Analyzing card in reader: Avtor KP375-BLE 0

--------------===========================--------------

Done.
CertUtil: -SCInfo command completed successfully.

TEST 2. Plugged in USB Smart Card reader, removed PIV card from BR301 and inserted into physical reader, then ran certutil -scinfo:

C:\Users\tperf>certutil -scinfo
The Microsoft Smart Card Resource Manager is running.
Current reader/card status:
Readers: 2
    0: Avtor KP375-BLE 0
    1: Gemalto USB SmartCard Reader 0
--- Reader: Avtor KP375-BLE 0
--- Status: SCARD_STATE_EMPTY
--- Status: No card.
---   Card:
--- Reader: Gemalto USB SmartCard Reader 0
--- Status: SCARD_STATE_PRESENT
--- Status: The card is available for use.
---   Card: Identity Device (NIST SP 800-73 [PIV])
---    ATR:
                3b fc 18 00 00 81 31 80  45 90 67 46 4a 00 64 16   ;.....1.E.gFJ.d.
                06 f2 72 7e 00 e0                                  ..r~..

=======================================================
Analyzing card in reader: Avtor KP375-BLE 0

--------------===========================--------------

=======================================================
Analyzing card in reader: Gemalto USB SmartCard Reader 0

--------------===========================--------------
================ Certificate 0 ================
--- Reader: Gemalto USB SmartCard Reader 0
---   Card: Identity Device (NIST SP 800-73 [PIV])
Provider = Microsoft Base Smart Card Crypto Provider
Key Container = aaab58a2-8a92-4de0-81d8-00f55d5fc101

No AT_SIGNATURE key for reader: Gemalto USB SmartCard Reader 0
Serial Number: 1658e989000100003179
Issuer: CN=PIVKey Device Certificate Authority, DC=pivkey, DC=com
 NotBefore: 4/26/2018 5:47 PM
 NotAfter: 4/23/2030 5:47 PM
Subject: CN=PIVKey A258ABAA928AE04D81D800F55D072F63
Non-root Certificate
Template: 1.3.6.1.4.1.311.21.8.8307090.11074004.13125291.8020695.16228079.167.15452091.6663618
Cert Hash(sha1): be4f704a67496c381fd2bea76b1636d73d0d6f81

Performing AT_KEYEXCHANGE public key matching test...
Public key matching test succeeded
    Key Container = aaab58a2-8a92-4de0-81d8-00f55d5fc101
    Provider = Microsoft Base Smart Card Crypto Provider
[... 115 lines cut ...]
Exclude leaf cert:
    Chain: da39a3ee5e6b4b0d3255bfef95601890afd80709
Full chain:
    Chain: be4f704a67496c381fd2bea76b1636d73d0d6f81
Missing Issuer: CN=PIVKey Device Certificate Authority, DC=pivkey, DC=com
    Issuer: CN=PIVKey Device Certificate Authority, DC=pivkey, DC=com
    NotBefore: 4/26/2018 5:47 PM
    NotAfter: 4/23/2030 5:47 PM
    Subject: CN=PIVKey A258ABAA928AE04D81D800F55D072F63
    Serial: 1658e989000100003179
    Template: 1.3.6.1.4.1.311.21.8.8307090.11074004.13125291.8020695.16228079.167.15452091.6663618
    Cert: be4f704a67496c381fd2bea76b1636d73d0d6f81
A certificate chain could not be built to a trusted root authority. 0x800b010a (-2146762486 CERT_E_CHAINING)
------------------------------------
Incomplete certificate chain
Cannot find certificate:
        CN=PIVKey Device Certificate Authority, DC=pivkey, DC=com
Displayed  cert for reader: Gemalto USB SmartCard Reader 0

--------------===========================--------------

Done.
CertUtil: -SCInfo command completed successfully.

C:\Users\tperf>

FeitianSmartcardReader, 2020-10-23:
Noted, the code only for you to have reference, anyway, would you please share your demo code to me, I forward to R&D team and do check, it would be better solve the issue based on your code.

Thnaks,

tperfitt, 2020-10-23:
I have been using a BLE app to write the values and get the responses before writing any code. I could write up some code to share, but I you couldn't really test it because you don't have the PIV cards that are not working. I seem to get the same response when I do this:

1. Sample Apps in Feitian github
2. Feitian App in App Store
3. Feitian CCID driver in Windows
4. BLE commands in an app

It seems to me to be an issue either with my PIV cards and the interaction with the reader. The same CCID commands seems to work fine with either the reader in USB mode or a different USB reader.

It should work with the CCID driver in Windows, correct? I can get it to pair, see the card and insertion events, but hangs on the certificates:

C:\Users\tperf>certutil -scinfo
The Microsoft Smart Card Resource Manager is running.
Current reader/card status:
Readers: 1
  0: Avtor KP375-BLE 0
--- Reader: Avtor KP375-BLE 0
--- Status: SCARD_STATE_PRESENT
--- Status: The card is available for use.
---   Card: Identity Device (NIST SP 800-73 [PIV])
---    ATR:
        3b fc 18 00 00 81 31 80  45 90 67 46 4a 00 64 16   ;.....1.E.gFJ.d.
        06 f2 72 7e 00 e0                                  ..r~..

=======================================================
Analyzing card in reader: Avtor KP375-BLE 0

--------------===========================--------------
================ Certificate 0 ================
--- Reader: Avtor KP375-BLE 0
---   Card: Identity Device (NIST SP 800-73 [PIV])
Provider = Microsoft Base Smart Card Crypto Provider
Key Container = (null) [Default Container]

Cannot open the AT_SIGNATURE key for reader: Avtor KP375-BLE 0
Cannot open the AT_KEYEXCHANGE key for reader: Avtor KP375-BLE 0

--------------===========================--------------
================ Certificate 0 ================
--- Reader: Avtor KP375-BLE 0
---   Card: Identity Device (NIST SP 800-73 [PIV])
Provider = Microsoft Smart Card Key Storage Provider
Key Container = (null) [Default Container]

I want to find any scenario over bluetooth where it can successfully read the card before coding up to make sure there isn't an incompatibility with the PIV cards and the reader over bluetooth. Do you have any suggestions on how I can make it work with Windows or other way to show communication with a PIV card?

FeitianSmartcardReader, 2020-10-23:
I have PIV card and works on our side, and some of our customer also use PIV card. so I confirm it supports.
But cannot reproduce your issue, that's the problem.
or you can let me know which app you are using on iOS? and then we follow your steps to reproduce the issue, and help on this.

I sent my test log with iReader(download from appstore), with the commands that you provided, all works.

tperfitt, 2020-10-24:
I received the new reader and it no longer gives the error and I am getting responses like I am expecting. Thanks for working through it with me!

closed 2020-10-24 by tperfitt