FeitianSmartcardReader/FEITIAN_MOBILE_READERS #20: Old blue br301 reader returns error on FTreader.poweron();

Anioz, 2020-11-25:
Hiya,

Do you know of any reasons the SDK would return error 612 when using the old blue br301 (bt3) cardreaders (best guess is fine too)?

  --- stack trace ---
com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerPowerOn][err : 612][com.ftsafe.readerScheme.FTException: err : 612]
	at com.ftsafe.readerScheme.FTReader.throwFTError(FTReader.java:566)
	at com.ftsafe.readerScheme.FTReader.readerPowerOn(FTReader.java:86)
	at crc646a6d9b4284f7b127.Handle.n_handleMessage(Native Method)
	at crc646a6d9b4284f7b127.Handle.handleMessage(Handle.java:53)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7356)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:915)
}

When using the black br301ble it works perfectly fine with your newest version of the sdk

FeitianSmartcardReader, 2020-11-25:
does bR301(Blue) has this issue with all card? if yes, it may hardware card slot issue or something inside the card slot, block the PINs.

otherwise, it may communication issue, could you please share your card ATR and communicate data between reader and card? there have debug library in our SDK, you can use it to catch the communication data.

Anioz, 2020-11-25:
Ok ill try to catch it, thank you for the fast reply

Anioz, 2020-11-25:
Ive updated my original post with more specific info, on iOS ive implimented the PC/SC directly (SCardConnect, etc), i also have a ir301-u lightning connection reader on that platform if you want me to test something more direct (w/o the bluetooth connection interference)

FeitianSmartcardReader, 2020-11-26:
Do you have firmware version of your bR301 Blue? you can download "smartcard reader" app from appsotre, and click the info button to get the firmware version.

And also, please send me your card ATR, then I can take simulate test on our side.

And the log, please send to hongbin@ftsafe.com, will take a look, thanks

Anioz, 2020-11-26:
I think you might be on to something, both the iR301-U and the old blue br301 aren't detected/working in the appstore smartcard reader/iReader and the Card Utilities app (iOS 14.2 iPhone XS, the black ble br301 works perfect in each of those apps)

Is there a way to update those firmwares to the latest? Would be a quick fix if that solves it 

I'll try to fetch the ATR & the log meanwhile

FeitianSmartcardReader, 2020-11-26:
Does your reader is OEM version? connect your reader, and then check setting->General->About, scroll down, you will see new device pop up, which under SEID, check the vendor name, if it is OEM reader, then you will need contact to brand owner.

Anioz, 2020-11-26:
Perfect its detected as Zetes, ill mark this issue solved, thank you very much for all the help

closed 2020-11-26 by Anioz