FeitianSmartcardReader/FEITIAN_MOBILE_READERS #23: Bluetooth not detecting or Working

MJ-BH, 2021-08-31:
I Have Issues using the new SDK with Android 10 and 11 , the Blutoth BLE4 note working , not detecting anything , and not throwing any Erorr And ome thing for the Apk Demo or the App in the play Store .

FeitianSmartcardReader, 2021-09-01:
Install App on iOS:
https://itunes.apple.com/us/app/smartcard-reader/id525954151?mt=8

Install App on Android:
PCSCLite: https://play.google.com/store/apps/details?id=com.ftsafe.pcscdemo&hl=en&gl=US 
Private API: https://play.google.com/store/apps/details?id=com.ftsafe.cardreader&hl=en&gl=US

FeitianSmartcardReader, 2021-09-01:
You can download app from google play store, thanks

MJ-BH, 2021-09-01:
I find that we Have to Add this Permessions : "    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>"
    "<uses-feature android:name="android.hardware.bluetooth" android:required="true"/>"
  "  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
in the Android Manifest and w should  be always allow when we Allow the Permession .

closed 2021-09-01 by MJ-BH