FeitianSmartcardReader/FEITIAN_MOBILE_READERS #25: Card reader error

otashjumaev, 2021-11-11:
when do these errors occur??

[ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][usb_recv error len:-1][com.ftsafe.usb.UsbException: usb_recv error len:-1]

[ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][null][#<ERROR the result should be I-block>]

[ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][null][#<ERROR bSeq not equal>]

[ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][null][#<ERROR this is error>]

how to solve them?

FeitianSmartcardReader, 2021-11-11:
it looks like the SDK cannot read data from USB.
I sent mail to you. Would you please check your mail? 

Suggestions:
1. download our test App:

Install App on Android, to play with it
PCSCLite: https://play.google.com/store/apps/details?id=com.ftsafe.pcscdemo&hl=en&gl=US
Private API: https://play.google.com/store/apps/details?id=com.ftsafe.cardreader&hl=en&gl=US
2. if your code have this error, it would be better share us your demo code, our engineer will take a lool
3. or we can have remote session, our engineer will debug in your app

And here is latest SDK:
The latest SDK: https://download.ftsafe.com/files/reader/product/Mobile_SDK_20211109.zip

otashjumaev, 2021-11-11:
Thank you, after updating to the latest version, it is working fine.

closed 2021-11-11 by otashjumaev