FeitianSmartcardReader/FEITIAN_MOBILE_READERS #26: Is the Feitian bR301 Bluetooth reader accessible through CryptoTokenKit on iOS 15?

minfrin, 2021-11-11:
I have a Feitian bR301 Bluetooth reader which I would like to integrate into an app running on iOS15.

Is the Feitian reader available through CryptoTokenKit on iOS?

If not, what driver/framework do I need to use to do this?

FeitianSmartcardReader, 2021-11-12:
We do envelop the operation interface to PC/SC API, you can find it in our SDK.
And to work with CryptoTokenKit, you may need develop an extension:
https://developer.apple.com/documentation/cryptotokenkit

One of our customer twocanoes did, you also can talk o them to using their extension, any questions, contact us anytime, 
https://twocanoes.com/products/ios/smart-card-utility-for-iphone-and-ipad/

boby1975, 2021-12-23:
Hi, **minfrin**
Did you integrate this reader through **CryptoTokenKit** on **iOS**?

I'm trying to do the same thing with this reader connected to my iOS device using standard Lightning to USB OTG Adapter and microUSB-USB cable. The first step is to get the corresponding smart card slot via **TKSmartCardSlotManager**. I've added the **com.apple.security.smartcard** entitlement but still get nil when trying to access the manager object. The console logs an error 

	[smartcard] ctk: connecting to slot registration server (<NSXPCConnection: 0x2800d8000> connection to service named com.apple.ctkd.slot-client) failed, 
	error Error Domain=NSCocoaErrorDomain Code=4097 "connection to service named com.apple.ctkd.slot-client" UserInfo={NSDebugDescription=connection to service named com.apple.ctkd.slot-client}

**Could you share your solution if possible?)**
Thnks.

tperfitt, 2022-01-06:
only devices approved for the MFI program can talk via the lightning port. The micro USB port does not have a MFI chip in it on the br301 so it can't talk to iOS. In fact, no USB readers are approved for MFI at this point. The device has to have a lightning connector to be approved.  For the br301, you have to communicate with it via the SDK over bluetooth for it to work.

Another option is to use the Feitian lightning readers. Those are MFI certified and work fine via the lightning port.

Finally, check out our app "Smart Card Utility" if your card is a PIV card. It should work fine with your reader.

tim