FeitianSmartcardReader/FEITIAN_MOBILE_READERS #27: /Android_SDK/new_android_sdk/Private_API/Demo code/ doesn't work with new FTReaderAPI_1.0.9.jar

boby1975, 2021-11-20:
When in "private API DemoCode" change FTReaderAPI_1.0.8.jar to FTReaderAPI_1.0.9.jar app crashes.
1) put file FTReaderAPI_1.0.9.jar in folder ..\Android_SDK\new_android_sdk\Private_API\Demo code\app\libs and delete old file FTReaderAPI_1.0.8.jar
2) change in file  ..\Android_SDK\new_android_sdk\Private_API\Demo code\app\build.gradle row "implementation files('libs/FTReaderAPI_1.0.8.jar')" to "implementation files('libs/FTReaderAPI_1.0.**9**.jar')"
3) build and run app
4) set device type BT4, press find, select found KP375BLE card reader (FT_xxxxx)
5) press power on in app and app crashes

debug info before crash:
A/safe.cardreade: java_vm_ext.cc:578] JNI DETECTED ERROR IN APPLICATION: JNI CallIntMethod called with pending exception java.lang.NoSuchMethodError: no non-static method "Lcom/ftsafe/comm/CommBase;.getDevType()I"
    java_vm_ext.cc:578]   at int com.ftsafe.ftnative.FTNative.readerPowerOn(int, byte[], int[], int) ((null):-2)
    java_vm_ext.cc:578]   at byte[] com.ftsafe.comm.CommBase.a(int, int) ((null):-1)
    java_vm_ext.cc:578]   at byte[] com.ftsafe.readerScheme.FTReader.readerPowerOn(int) ((null):-1)
    java_vm_ext.cc:578]   at void com.ftsafe.cardreader.MainActivity$6.run() (MainActivity.java:433)
    java_vm_ext.cc:578]   at void com.ftsafe.cardreader.MainActivity$20.run() (MainActivity.java:720)
    java_vm_ext.cc:578]   at void java.lang.Thread.run() (Thread.java:923)
    java_vm_ext.cc:578] 
    java_vm_ext.cc:578]     in call to CallIntMethod
    java_vm_ext.cc:578]     from int com.ftsafe.ftnative.FTNative.readerPowerOn(int, byte[], int[], int)
...
A/safe.cardreade: runtime.cc:677] Runtime aborting...
    runtime.cc:677] Dumping all threads without mutator lock held
    runtime.cc:677] All threads:
    runtime.cc:677] DALVIK THREADS (98):
    runtime.cc:677] "Thread-305" prio=5 tid=90 Runnable
    runtime.cc:677]   | group="" sCount=0 dsCount=0 flags=0 obj=0x12f0ef28 self=0x72fd2e5800
    runtime.cc:677]   | sysTid=3192 nice=0 cgrp=default sched=0/0 handle=0x72973b8cc0
    runtime.cc:677]   | state=R schedstat=( 12118961 183616 11 ) utm=0 stm=0 core=7 HZ=100
    runtime.cc:677]   | stack=0x72972b5000-0x72972b7000 stackSize=1043KB
    runtime.cc:677]   | held mutexes= "abort lock" "mutator lock"(shared held)
    runtime.cc:677]   native: #00 pc 00000000004a7ee8  /apex/com.android.art/lib64/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+140)
    runtime.cc:677]   native: #01 pc 00000000005b59dc  /apex/com.android.art/lib64/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool, BacktraceMap*, bool) const+376)
    runtime.cc:677]   native: #02 pc 00000000005d2b14  /apex/com.android.art/lib64/libart.so (art::DumpCheckpoint::Run(art::Thread*)+924)
    runtime.cc:677]   native: #03 pc 00000000005cca54  /apex/com.android.art/lib64/libart.so (art::ThreadList::RunCheckpoint(art::Closure*, art::Closure*)+528)
    runtime.cc:677]   native: #04 pc 00000000005cbc20  /apex/com.android.art/lib64/libart.so (art::ThreadList::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)+1920)
    runtime.cc:677]   native: #05 pc 000000000056656c  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort(char const*)+1864)
    runtime.cc:677]   native: #06 pc 0000000000013978  /system/lib64/libbase.so (android::base::SetAborter(std::__1::function<void (char const*)>&&)::$_3::__invoke(char const*)+76)
    runtime.cc:677]   native: #07 pc 0000000000012fa4  /system/lib64/libbase.so (android::base::LogMessage::~LogMessage()+320)
    runtime.cc:677]   native: #08 pc 000000000038e1d4  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+2572)
    runtime.cc:677]   native: #09 pc 000000000038e24c  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbortV(char const*, char const*, std::__va_list)+108)
    runtime.cc:677]   native: #10 pc 000000000037ff2c  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::AbortF(char const*, ...)+144)
    runtime.cc:677]   native: #11 pc 000000000037ea44  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::CheckPossibleHeapValue(art::ScopedObjectAccess&, char, art::(anonymous namespace)::JniValueType)+3108)
    runtime.cc:677]   native: #12 pc 000000000037d6a4  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::Check(art::ScopedObjectAccess&, bool, char const*, art::(anonymous namespace)::JniValueType*)+612)
    runtime.cc:677]   native: #13 pc 0000000000382e48  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::CheckCallArgs(art::ScopedObjectAccess&, art::(anonymous namespace)::ScopedCheck&, _JNIEnv*, _jobject*, _jclass*, _jmethodID*, art::InvokeType, art::(anonymous namespace)::VarArgs const*)+132)
    runtime.cc:677]   native: #14 pc 0000000000381d94  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::CallMethodV(char const*, _JNIEnv*, _jobject*, _jclass*, _jmethodID*, std::__va_list, art::Primitive::Type, art::InvokeType)+728)
    runtime.cc:677]   native: #15 pc 000000000036ff20  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::CallIntMethod(_JNIEnv*, _jobject*, _jmethodID*, ...)+152)
    runtime.cc:677]   native: #16 pc 000000000000431c  /data/app/~~oja4l9HTxVEsYi3m3CmD7g==/com.ftsafe.cardreader-tQlw1cxSRdvVI66WlhUj-Q==/lib/arm64/libFTReaderPCSC_1.0.8.so (getDeviceType+280)
    runtime.cc:677]   native: #17 pc 0000000000009a7c  /data/app/~~oja4l9HTxVEsYi3m3CmD7g==/com.ftsafe.cardreader-tQlw1cxSRdvVI66WlhUj-Q==/lib/arm64/libFTReaderPCSC_1.0.8.so (CmdPowerOn+68)
    runtime.cc:677]   native: #18 pc 00000000000032b8  /data/app/~~oja4l9HTxVEsYi3m3CmD7g==/com.ftsafe.cardreader-tQlw1cxSRdvVI66WlhUj-Q==/lib/arm64/libFTReaderPCSC_1.0.8.so (readerPowerOn(_JNIEnv*, _jobject*, int, _jbyteArray*, _jintArray*, int)+184)
    runtime.cc:677]   at com.ftsafe.ftnative.FTNative.readerPowerOn(Native method)
    runtime.cc:677]   at com.ftsafe.comm.CommBase.a(unavailable:-1)
    runtime.cc:677]   at com.ftsafe.readerScheme.FTReader.readerPowerOn(unavailable:-1)
    runtime.cc:677]   at com.ftsafe.cardreader.MainActivity$6.run(MainActivity.java:433)
    runtime.cc:677]   at com.ftsafe.cardreader.MainActivity$20.run(MainActivity.java:720)
    runtime.cc:677]   at java.lang.Thread.run(Thread.java:923)
...

FeitianSmartcardReader, 2021-11-22:
@boby1975 https://download.ftsafe.com/files/reader/TH/CardReader.zip here is our latest demo.

boby1975, 2021-11-22:
Thanks a lot.
But problem is the same (
I have install existed app-releas.apk from archive folder CardReader/app/release, device type - BT4, find, select reader, press button "poweron" - app crashes.

info from Logcat
2021-11-22 11:09:23.667 15790-17476/com.ftsafe.cardreader A/safe.cardreade: java_vm_ext.cc:578] JNI DETECTED ERROR IN APPLICATION: JNI CallIntMethod called with pending exception java.lang.NoSuchMethodError: no non-static method "Lcom/ftsafe/comm/CommBase;.getDevType()I"
    java_vm_ext.cc:578]   at int com.ftsafe.ftnative.FTNative.readerPowerOn(int, byte[], int[], int) ((null):-2)
    java_vm_ext.cc:578]   at byte[] com.ftsafe.comm.CommBase.a(int, int) ((null):-1)
    java_vm_ext.cc:578]   at byte[] com.ftsafe.readerScheme.FTReader.readerPowerOn(int) ((null):-1)
    java_vm_ext.cc:578]   at void com.ftsafe.cardreader.MainActivity$6.run() (MainActivity.java:420)
    java_vm_ext.cc:578]   at void com.ftsafe.cardreader.MainActivity$20.run() (MainActivity.java:707)
    java_vm_ext.cc:578]   at void java.lang.Thread.run() (Thread.java:923)
    java_vm_ext.cc:578] 
    java_vm_ext.cc:578]     in call to CallIntMethod
    java_vm_ext.cc:578]     from int com.ftsafe.ftnative.FTNative.readerPowerOn(int, byte[], int[], int)
2021-11-22 11:09:23.966 15790-17476/com.ftsafe.cardreader A/safe.cardreade: runtime.cc:677] Runtime aborting...
    runtime.cc:677] Dumping all threads without mutator lock held
....

more over app crashes when device type - BT4, find, select reader,  then press button "get firmware version" - will be crash.

Logcat
2021-11-22 11:23:08.336 16221-18563/com.ftsafe.cardreader A/safe.cardreade: java_vm_ext.cc:578] JNI DETECTED ERROR IN APPLICATION: JNI CallIntMethod called with pending exception java.lang.NoSuchMethodError: no non-static method "Lcom/ftsafe/comm/CommBase;.getDevType()I"
    java_vm_ext.cc:578]   at int com.ftsafe.ftnative.FTNative.readerGetFirmwareVersion(byte[], int[]) ((null):-2)
    java_vm_ext.cc:578]   at byte[] com.ftsafe.comm.CommBase.f() ((null):-1)
    java_vm_ext.cc:578]   at java.lang.String com.ftsafe.readerScheme.FTReader.readerGetFirmwareVersion() ((null):-1)
    java_vm_ext.cc:578]   at void com.ftsafe.cardreader.MainActivity$13.run() (MainActivity.java:586)
    java_vm_ext.cc:578]   at void com.ftsafe.cardreader.MainActivity$20.run() (MainActivity.java:707)
    java_vm_ext.cc:578]   at void java.lang.Thread.run() (Thread.java:923)
    java_vm_ext.cc:578] 
    java_vm_ext.cc:578]     in call to CallIntMethod
    java_vm_ext.cc:578]     from int com.ftsafe.ftnative.FTNative.readerGetFirmwareVersion(byte[], int[])
2021-11-22 11:23:08.559 16221-18563/com.ftsafe.cardreader A/safe.cardreade: runtime.cc:677] Runtime aborting...

PS
I guess it is due obfuscation of FTReaderAPI_1.0.9.jar

FeitianSmartcardReader, 2022-03-10:
fixed in 1.0.9.4.

closed 2022-03-10 by FeitianSmartcardReader