FeitianSmartcardReader/FEITIAN_MOBILE_READERS #28: [ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][err : 612][com.ftsafe.readerScheme.FTException: err : 612]

otashjumaev, 2021-12-02:
hello, I'm getting this error on some devices,
 can you help me?

here is the whole stacktrace:

```
Exception: com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][err : 612]
[com.ftsafe.readerScheme.FTException: err : 612]
	at com.ftsafe.readerScheme.FTReader.throwFTError(FTReader.java:566)
	at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:133)
	at uz.newtech.mobilkassa.ui.activities.MainActivity$sendCommands$2.invokeSuspend(MainActivity.kt:522)
	at uz.newtech.mobilkassa.ui.activities.MainActivity$sendCommands$2.invoke(Unknown Source)
	at uz.newtech.mobilkassa.ui.activities.MainActivity$sendCommands$2.invoke(Unknown Source)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:89)
	at kotlinx.coroutines.CoroutineScopeKt.coroutineScope(CoroutineScope.kt:264)
	at uz.newtech.mobilkassa.ui.activities.MainActivity.sendCommands(MainActivity.kt:515)
	at uz.newtech.mobilkassa.ui.activities.MainActivity$sendCommand$1.invokeSuspend(MainActivity.kt:703)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.internal.ScopeCoroutine.afterResume(Scopes.kt:33)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:102)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
```

FeitianSmartcardReader, 2021-12-09:
@otashjumaev we will take a look on tomorrow

FeitianSmartcardReader, 2021-12-09:
from the log, it looks communication issue, would you please send mail to me hongbin@ftsafe.com

And also, please check your reader's firmware version. The attached screenshot shows you how to get the firmware version.
[image]
n

FeitianSmartcardReader, 2022-03-10:
please using 1.0.9.4, our latest lib and demo code.https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/files/8219836/CardReader_SDK_For_Android_1.0.9.4_20211221.zip

closed 2022-03-10 by FeitianSmartcardReader