FeitianSmartcardReader/FEITIAN_MOBILE_READERS #31: CRASH BUG IN FTReaderAPI, in method com.ftsafe.comm.bt4.BT4.BlueToothRead() -  OutOfMemoryError

boby1975, 2021-12-19:
Crash occurs in method "**com.ftsafe.comm.bt4.BT4.BlueToothRead()**" when reading certs from card with BLE reader in my app using PCSC API or private API (doesn't matter in both cases). It is rare but has already met several times. Sometimes we catch this exemption "**com.ftsafe.comm.bt4.Bt4Exception: BlueToothRead:getPipeByte timeout**" without crash. But sometimes there is such exemption with warning like this "**Throwing OutOfMemoryError "Failed to allocate a 2030055072 byte allocation with 2957247 free bytes and 189MB until OOM**" and with crash.

As I understand sometimes there is attempt to allocate a huge size of memory for **readDataBuff** in method "**com.ftsafe.comm.bt4.BT4.BlueToothRead**". May be here should done some validation before memory allocation. 

**I thing it's impossible to fix this on my app side, only in libs it's possible to fix.**

**info from logcat when crash occurs:**

	2021-12-19 16:20:26.912 2101-4917/com.androidfeitianfirstapp.my E/FTSAFE: [BT4-SEND]6F05000000006A000000803200000A
	2021-12-19 16:20:26.917 2101-4917/com.androidfeitianfirstapp.my E/FTSAFE: timeout:3000
	2021-12-19 16:20:26.918 2101-2986/com.androidfeitianfirstapp.my E/FTSAFE: [BLE-onCharacteristicWrite]6F05000000006A000000803200000A
	2021-12-19 16:20:26.983 2101-2986/com.androidfeitianfirstapp.my E/FTSAFE: [BLE-onCharacteristicChanged]800C000000006A0000001FCC862E007900270004
	2021-12-19 16:20:26.984 2101-2986/com.androidfeitianfirstapp.my E/FTSAFE: NEW:800c000000006a0000001fcc862e007900270004
	2021-12-19 16:20:26.985 2101-2986/com.androidfeitianfirstapp.my E/FTSAFE: [BLE-onCharacteristicChanged]9000
	2021-12-19 16:20:26.985 2101-2986/com.androidfeitianfirstapp.my E/FTSAFE: NEW:9000
	2021-12-19 16:20:29.991 2101-4917/com.androidfeitianfirstapp.my W/System.err: com.ftsafe.comm.bt4.Bt4Exception: BlueToothRead:getPipeByte timeout
	2021-12-19 16:20:29.993 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at com.ftsafe.comm.bt4.BT4.BlueToothRead(BT4.java:595)
	2021-12-19 16:20:29.995 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at com.ftsafe.comm.bt4.BT4.ft_recv(BT4.java:375)
	2021-12-19 16:20:29.997 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at com.ftsafe.comm.CommBase.readData(CommBase.java:110)
	2021-12-19 16:20:29.999 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at sun.security.pkcs11.wrapper.PKCS11.C_FindObjectsInit(Native Method)
	2021-12-19 16:20:30.000 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at ua.avtor.DsLib.PrivateKeys.PKCS11Utils.Pkcs11Session.findAllObjects(Unknown Source:4)
	2021-12-19 16:20:30.001 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at ua.avtor.DsLib.PrivateKeys.PKCS11Utils.Pkcs11Session.findAllObjects(Unknown Source:13)
	2021-12-19 16:20:30.002 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at ua.avtor.DsLib.KeyStores.KeyStorePKCS11.ensureCertificates(Unknown Source:42)
	2021-12-19 16:20:30.003 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at ua.avtor.DsLib.KeyStores.KeyStorePKCS11.getCertificates(Unknown Source:0)
	2021-12-19 16:20:30.005 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at com.androidfeitianfirstapp.avmacbook.androidfeitianfirstapp.MainActivity$1$1.run(MainActivity.java:89)
	2021-12-19 16:20:30.005 2101-4917/com.androidfeitianfirstapp.my W/System.err:     at java.lang.Thread.run(Thread.java:923)
	2021-12-19 16:20:30.006 2101-4917/com.androidfeitianfirstapp.my E/FTSAFE: timeout:3000
	2021-12-19 16:20:30.079 2101-4917/com.androidfeitianfirstapp.my W/tianfirstapp.m: Throwing OutOfMemoryError "Failed to allocate a 2030055068 byte allocation with 2967462 free bytes and 189MB until OOM, target footprint 5934926, growth limit 201326592" (VmSize 5841512 kB)
	2021-12-19 16:20:30.138 2101-4917/com.androidfeitianfirstapp.my W/tianfirstapp.m: Throwing OutOfMemoryError "Failed to allocate a 2030055072 byte allocation with 2957247 free bytes and 189MB until OOM, target footprint 5914495, growth limit 201326592" (VmSize 5841512 kB)
	2021-12-19 16:20:30.139 2101-4917/com.androidfeitianfirstapp.my A/tianfirstapp.m: java_vm_ext.cc:578] JNI DETECTED ERROR IN APPLICATION: JNI GetByteArrayElements called with pending exception java.lang.OutOfMemoryError: Failed to allocate a 2030055072 byte allocation with 2957247 free bytes and 189MB until OOM, target footprint 5914495, growth limit 201326592
    java_vm_ext.cc:578]   at byte[] com.ftsafe.comm.bt4.BT4.BlueToothRead(int) (BT4.java:584)
    java_vm_ext.cc:578]   at byte[] com.ftsafe.comm.bt4.BT4.ft_recv(int, int) (BT4.java:375)
    java_vm_ext.cc:578]   at int com.ftsafe.comm.CommBase.readData(int, int, byte[], int[]) (CommBase.java:110)
    java_vm_ext.cc:578]   at void sun.security.pkcs11.wrapper.PKCS11.C_FindObjectsInit(long, sun.security.pkcs11.wrapper.CK_ATTRIBUTE[]) (PKCS11.java:-2)
    java_vm_ext.cc:578]   at long[] ua.avtor.DsLib.PrivateKeys.PKCS11Utils.Pkcs11Session.findAllObjects(sun.security.pkcs11.wrapper.CK_ATTRIBUTE[]) ((null):-1)
    java_vm_ext.cc:578]   at long[] ua.avtor.DsLib.PrivateKeys.PKCS11Utils.Pkcs11Session.findAllObjects(long) ((null):-1)
    java_vm_ext.cc:578]   at void ua.avtor.DsLib.KeyStores.KeyStorePKCS11.ensureCertificates() ((null):-1)
    java_vm_ext.cc:578]   at ua.avtor.DsLib.Certificate[] ua.avtor.DsLib.KeyStores.KeyStorePKCS11.getCertificates(int, boolean, int, java.lang.String) ((null):-1)
    java_vm_ext.cc:578]   at void com.androidfeitianfirstapp.avmacbook.androidfeitianfirstapp.MainActivity$1$1.run() (MainActivity.java:89)

FeitianSmartcardReader, 2021-12-20:
our engineer working on it, keep update, thanks

boby1975, 2021-12-22:
As I understand **CCID answer can't be more than 65,538 bytes**.

*************************************************************************************************************************************
According official documentation
[https://usb.org/sites/default/files/DWG_Smart-Card_CCID_Rev110.pdf](https://usb.org/sites/default/files/DWG_Smart-Card_CCID_Rev110.pdf)

**6.2.1 RDR_to_PC_DataBlock**

**dwLength** - Size of abData field of this message

**abData** - This field contains the data returned by the CCID. Depending on the exchange level, this may be data and status “as is” from the ICC, or the CCID may “filter” the data and status before sending it to the host. **It can be 0 - 65,538 bytes in length**
*************************************************************************************************************************************

It may be worth **limiting** the buffer size of **readDataBuff** in **com.ftsafe.comm.bt4.BT4.BlueToothRead** to this value (65,538 + 10 = **65548**)

boby1975, 2022-01-04:
We again caught such crash in latest **FTReaderAPI_1.0.9.4.jar**.

**info from logcat when crash occurs at 2022-01-04 12:22:39.525:**

	2022-01-04 12:22:35.749 22012-32339/ua.avtor.umcaservice.debug I/JNI PCSC_Wrapper: JNI_SCardTransmit
	2022-01-04 12:22:35.895 22012-32339/ua.avtor.umcaservice.debug E/FT: recvHead===============:800C0000000073000000
	2022-01-04 12:22:35.896 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_len==============:12
	2022-01-04 12:22:35.896 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_readDataBuffLen==============:22
	2022-01-04 12:22:35.899 22012-32339/ua.avtor.umcaservice.debug I/JNI PCSC_Wrapper: JNI_SCardTransmit
	2022-01-04 12:22:36.045 22012-32339/ua.avtor.umcaservice.debug E/FT: recvHead===============:800C0000000074000000
	2022-01-04 12:22:36.045 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_len==============:12
	2022-01-04 12:22:36.045 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_readDataBuffLen==============:22
	2022-01-04 12:22:36.049 22012-32339/ua.avtor.umcaservice.debug I/JNI PCSC_Wrapper: JNI_SCardTransmit
	2022-01-04 12:22:36.272 22012-32339/ua.avtor.umcaservice.debug E/FT: recvHead===============:800C0000000075000000
	2022-01-04 12:22:36.272 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_len==============:12
	2022-01-04 12:22:36.272 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_readDataBuffLen==============:22
	2022-01-04 12:22:39.280 22012-32339/ua.avtor.umcaservice.debug W/System.err: com.ftsafe.comm.bt4.Bt4Exception: BlueToothRead:getPipeByte timeout
	2022-01-04 12:22:39.281 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at com.ftsafe.comm.bt4.BT4.BlueToothRead(BT4.java:599)
	2022-01-04 12:22:39.282 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at com.ftsafe.comm.bt4.BT4.ft_recv(BT4.java:378)
	2022-01-04 12:22:39.283 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at com.ftsafe.comm.CommBase.readData(CommBase.java:114)
	2022-01-04 12:22:39.284 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at com.ftsafe.ftnative.FTNative.readerXfrBlock(Native Method)
	2022-01-04 12:22:39.285 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at com.ftsafe.comm.CommBase.readerXfrBlock(CommBase.java:394)
	2022-01-04 12:22:39.286 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:126)
	2022-01-04 12:22:39.287 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at avtor.kp375blemodule.KP375BLE_Module.SCardTransmit(KP375BLE_Module.kt:372)
	2022-01-04 12:22:39.288 22012-32339/ua.avtor.umcaservice.debug W/System.err:     at avtor.pcscmodule.PCSC_Module.SCardTransmit(PCSC_Module.kt:164)
	2022-01-04 12:22:39.289 22012-32339/ua.avtor.umcaservice.debug E/FT: recvHead===============:800000021BE400760001
	2022-01-04 12:22:39.289 22012-32339/ua.avtor.umcaservice.debug E/FT: BT4_BlueToothRead()_len==============:453115904
	2022-01-04 12:22:39.290 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Starting a blocking GC Alloc
	2022-01-04 12:22:39.290 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Starting a blocking GC Alloc
	2022-01-04 12:22:39.336 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Alloc young concurrent copying GC freed 50069(1564KB) AllocSpace objects, 19(1292KB) LOS objects, 46% free, 5767KB/10MB, paused 213us total 45.268ms
	2022-01-04 12:22:39.336 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Forcing collection of SoftReferences for 432MB allocation
	2022-01-04 12:22:39.336 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Starting a blocking GC Alloc
	2022-01-04 12:22:39.410 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Alloc concurrent copying GC freed 3116(171KB) AllocSpace objects, 1(68KB) LOS objects, 49% free, 5575KB/10MB, paused 768us total 73.249ms
	2022-01-04 12:22:39.411 22012-32339/ua.avtor.umcaservice.debug W/caservice.debu: Throwing OutOfMemoryError "Failed to allocate a 453115926 byte allocation with 5709750 free bytes and 186MB until OOM, target footprint 11419502, growth limit 201326592" (VmSize 25845216 kB)
	2022-01-04 12:22:39.412 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Starting a blocking GC Alloc
	2022-01-04 12:22:39.412 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Starting a blocking GC Alloc
	2022-01-04 12:22:39.451 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Alloc young concurrent copying GC freed 153(47KB) AllocSpace objects, 0(0B) LOS objects, 49% free, 5576KB/10MB, paused 296us total 38.568ms
	2022-01-04 12:22:39.451 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Forcing collection of SoftReferences for 432MB allocation
	2022-01-04 12:22:39.451 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Starting a blocking GC Alloc
	2022-01-04 12:22:39.523 22012-32339/ua.avtor.umcaservice.debug I/caservice.debu: Alloc concurrent copying GC freed 244(40KB) AllocSpace objects, 0(0B) LOS objects, 49% free, 5568KB/10MB, paused 359us total 71.945ms
	2022-01-04 12:22:39.523 22012-32339/ua.avtor.umcaservice.debug W/caservice.debu: Throwing OutOfMemoryError "Failed to allocate a 453115928 byte allocation with 5701959 free bytes and 186MB until OOM, target footprint 11403919, growth limit 201326592" (VmSize 25845216 kB)
[... cut ...] tor.pcscmodule.struct.LPDWORD) (PCSC_Module.kt:164)
    	java_vm_ext.cc:578] 
    	java_vm_ext.cc:578]     in call to GetByteArrayElements
    	java_vm_ext.cc:578]     from int com.ftsafe.ftnative.FTNative.readerXfrBlock(int, byte[], int, byte[], int[])

As we can see at **2022-01-04 12:22:39.289** header from smartcard is OK (**800000021BE400760001**).
But in method com.ftsafe.comm.bt4.BT4.**BlueToothRead** there is attempt to allocate a huge size (**453115904**) of memory for **readDataBuff**.

boby1975, 2022-01-05:
We have investigated this problem more deeply. Allocation  of memory for **readDataBuff** in method "**com.ftsafe.comm.bt4.BT4.BlueToothRead**" only depends on raw data from smartcard. But theoretically raw data of smartcard can content mistake in some rare cases. More over, what we have actually received in **recvHead** buffer in method **BlueToothRead** can be in some rare cases not the same what smartcard have sent really. Thus, we need to do some check of calculated value **len** of the **readDataBuff** buffer size for receiving data from a smartcard.

As mentioned [above](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/31#issuecomment-999488586) CCID answer **can't be more than 65,538 bytes**.

### Thus, how to fix:
**solution#1**
in method "**com.ftsafe.comm.bt4.BT4.BlueToothRead**" after row

    int len = (recvHead[1] & 255) + ((recvHead[2] & 255) << 8) + ((recvHead[3] & 255) << 16) + ((recvHead[4] & 255) << 24);

add row

    if ( len > 65538 ) len = 65538;

This prevents "**Out of memory crash**".

boby1975, 2022-01-10:
This problem  was fixed in **FTReaderAPI_1.0.9.5.jar**. Thanks.

Now in such rare cases we have limitation for **readDataBuff** size without crash. We caught such a case.

    E/FT: recvHead===============:8B330064002700049000
    BT4_BlueToothRead()_len==============:65538
    BT4_BlueToothRead()_readDataBuffLen==============:65548

**FYI**
It is noticed that **always** before such cases there is a caught exception

    W/System.err: com.ftsafe.comm.bt4.Bt4Exception: BlueToothRead:getPipeByte timeout
    W/System.err:     at com.ftsafe.comm.bt4.BT4.BlueToothRead(BT4.java:601)

But it's "another story" and need more deep investigation why this happens.

FeitianSmartcardReader, 2022-01-11:
Do you have video or demo source code allow us to reproduce the issue? @boby1975 our engineer cannot reproduce it, so if we can reproduce it, it would be fast to solve on our side, thanks and looking forward your reply,

boby1975, 2022-01-11:
Unfortunately there is no video. Such **"Exception: BlueToothRead:getPipeByte timeout"**  sometimes occurs when reading certificates from a smartcard by our library PKCS11 when there is a long series of call method "**FTReader.readerXfr**" in SCardTransmit of our PCSC. It takes about 15-20 seconds.

	I/JNI PCSC_Wrapper: JNI_SCardTransmit
	E/FT: readerXfr_index ====== :0
	I/JNI PCSC_Wrapper: JNI_SCardTransmit
	E/FT: readerXfr_index ====== :0
	...
	I/JNI PCSC_Wrapper: JNI_SCardTransmit
	E/FT: readerXfr_index ====== :0
	W/System.err: com.ftsafe.comm.bt4.Bt4Exception: BlueToothRead:getPipeByte timeout

Of course it is difficult to reproduce when it appears occasionally and there is no systematic nature of its occurrence.

YoeriJPL, 2022-03-11:
We seem to be having this issue as well on Android:
03-11 12:42:02.936	Samsung SM-G935F	Error	13875	DEBUG	Abort message: 'JNI DETECTED ERROR IN APPLICATION: JNI GetByteArrayElements called with pending exception java.lang.OutOfMemoryError: Failed to allocate a 674247032 byte allocation with 25165824 free bytes and 187MB until OOM, target footprint 29643736, growth limit 201326592
  at byte[] com.ftsafe.comm.bt4.BT4.BlueToothRead(int) (BT4.java:403)
  at byte[] com.ftsafe.comm.bt4.BT4.ft_recv(int, int) (BT4.java:337)
  at int com.ftsafe.comm.CommBase.readData(int, int, byte[], int[]) (CommBase.java:110)
  at int com.ftsafe.ftnative.FTNative.readerXfrBlock(int, byte[], int, byte[], int[]) (FTNative.java:-2)
  at byte[] com.ftsafe.comm.CommBase.readerXfrBlock(int, byte[]) (CommBase.java:373)
  at byte[] com.ftsafe.readerScheme.FTReader.readerXfr(int, byte[]) (FTReader.java:126)
03-11 12:42:02.120	Samsung SM-G935F	Error	12866	.client.androi	java_vm_ext.cc:579] JNI DETECTED ERROR IN APPLICATION: JNI GetByteArrayElements called with pending exception java.lang.OutOfMemoryError: Failed to allocate a 674247032 byte allocation with 25165824 free bytes and 187MB until OOM, target footprint 29643736, growth limit 201326592

It causes a full crash of our app even tough there are try catches around the methods

boby1975, 2022-03-12:
try to use **FTReaderAPI_1.0.9.5.jar**. it prevents from crash.

FeitianSmartcardReader, 2022-03-14:
[CardReader_1.0.9.5.zip](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/files/8241680/CardReader_1.0.9.5.zip)
@YoeriJPL here is 1.0.9.5

FeitianSmartcardReader, 2022-03-14:
follow the boby1975 comments:
https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/31#issuecomment-1005565229 
avoid the crash issue

YoeriJPL, 2022-03-14:
We've succesfully updated to FTReaderAPI_1.0.9.5.jar and produced a test version for our Android users, currently test are being done but quick tests show reliability was at a good level

If available: Can you send us an updatet version of the iOS side drivers for testing as well?

Sincere thanks,

FeitianSmartcardReader, 2022-03-15:
@YoeriJPL here is our latest 3.5.64 version, new version improve the card slot event for BLE reader only.
https://docs.ftsafe.cn/s/kWor9fK5A8RQ8eE

YoeriJPL, 2022-03-15:
The error Outofmemory doesn't seem to happen anymore, the app no longer crashes either but the notorious getpipebyte error mentioned before pops up every now and then and prevents readouts when it happens. Switching the cardreader off and back on fixes the problem for the user as a workaround:

[FT] onResult=======================128
[FT] recvHead===============:800D0000000001000000
[FT] BT4_BlueToothRead()_len==============:13
[FT] BT4_BlueToothRead()_readDataBuffLen==============:23
[FT] onResult=======================128
Thread started: <Thread Pool> #40
[FT] onResult=======================128
[FT] onResult=======================128
[FT] recvHead===============:800D0000000002000000
[FT] BT4_BlueToothRead()_len==============:13
[FT] BT4_BlueToothRead()_readDataBuffLen==============:23
[FT] onResult=======================128
[FT] onResult=======================128
[FT] onResult=======================128
[FT] recvHead===============:82050000000003000000
[FT] BT4_BlueToothRead()_len==============:5
[FT] BT4_BlueToothRead()_readDataBuffLen==============:15
[FT] onResult=======================128
[BluetoothGatt] onConnectionUpdated() - Device=C1:5A:92:19:C7:B2 interval=60 latency=0 timeout=400 status=0
[System.err] com.ftsafe.comm.bt4.Bt4Exception: BlueToothRead:getPipeByte timeout
[System.err] 	at com.ftsafe.comm.bt4.BT4.BlueToothRead(BT4.java:601)
[System.err] 	at com.ftsafe.comm.bt4.BT4.ft_recv(BT4.java:378)
[System.err] 	at com.ftsafe.comm.CommBase.readData(CommBase.java:119)
[System.err] 	at com.ftsafe.ftnative.FTNative.readerPowerOn(Native Method)
[System.err] 	at com.ftsafe.comm.CommBase.readerPowerOn(CommBase.java:379)
[System.err] 	at com.ftsafe.readerScheme.FTReader.readerPowerOn(FTReader.java:85)
[.client.androi] Long monitor contention with owner Thread-19 (18938) at int com.ftsafe.comm.CommBase.readData(int, int, byte[], int[])(CommBase.java:-1) waiters=0 in int com.ftsafe.comm.CommBase.writeData(int, byte[], int) for 2.924s

YoeriJPL, 2022-03-15:
On Android Q/10 driver v1.0.9.5 we have caught an error that might or might not be related to the above error:
Com.Ftsafe.ReaderScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerOpen][mBluetoothLeClass.connect failed][com.ftsafe.comm.bt4.Bt4Exception: mBluetoothLeClass.connect failed]
  at Java.Interop.JniEnvironment+InstanceMethods.CallObjectMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) <0x7c9bdd69e0 + 0x0012f> in <00c315a988634383b446eff646084784>:0 
  at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeVirtualObjectMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) <0x7c9bdddc60 + 0x000df> in <00c315a988634383b446eff646084784>:0 
  at Com.Ftsafe.ReaderScheme.FTReader.ReaderOpen (Java.Lang.Object device) [0x00031] in <00493e34c83a424bb5e5cbd6b1bb71dd>:0 
  at C4nMobile.Droid.Dependency.Handle.ConnectProcedure (System.Boolean IsFavReaderKnown, Android.Bluetooth.BluetoothDevice device) [0x000c0] in <bf37bb79aead4b8aade8560d6ac77390>:0 
  --- End of managed Com.Ftsafe.ReaderScheme.FTException stack trace ---
com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerOpen][mBluetoothLeClass.connect failed][com.ftsafe.comm.bt4.Bt4Exception: mBluetoothLeClass.connect failed]
	at com.ftsafe.readerScheme.FTReader.throwFTError(FTReader.java:580)
	at com.ftsafe.readerScheme.FTReader.readerOpen(FTReader.java:44)

Killing the app and restarting it had no effect, rebooting the device restored cardreader functionality

YoeriJPL, 2022-03-15:
thank you boby

YoeriJPL, 2022-03-16:
> @YoeriJPL here is our latest 3.5.64 version, new version improve the card slot event for BLE reader only. https://docs.ftsafe.cn/s/kWor9fK5A8RQ8eE

Has obfuscation been increased in this driver? We are trying to load it into the Xamarin iOS binding library and are failing to be able to access the classes on this version

YoeriJPL, 2022-03-17:
> > @YoeriJPL here is our latest 3.5.64 version, new version improve the card slot event for BLE reader only. https://docs.ftsafe.cn/s/kWor9fK5A8RQ8eE
> 
> Has obfuscation been increased in this driver? We are trying to load it into the Xamarin iOS binding library and are failing to access the classes on this version

The solution was to add Frameworks = "ExternalAccessory") into the standard xamarin ios binding linkwith.cs file

FeitianSmartcardReader, 2022-03-25:
noted with thanks,

closed 2022-03-25 by FeitianSmartcardReader