FeitianSmartcardReader/FEITIAN_MOBILE_READERS #32: PCSC API, FT_SCardListReaders always returns SCARD_E_NO_READERS_AVAILABLE after FT_SCardConnect is called

boby1975, 2021-12-20:
After calling method **FT_SCardConnect** (return SCARD_S_SUCCESS) the method **FT_SCardListReaders** always returns 0x8010002E (**SCARD_E_NO_READERS_AVAILABLE**), but BLE reader is connected (**blue and green leds are on**).
Is it correct behavior? 
**Why if YES?**

### How to reproduce:
**1)** open  project FEITIAN_MOBILE_READERS-master1.0.9.2\Android_SDK\new_android_sdk\PCSC_API\Demo code\pcscdemo
**2)** unhide (android:visibility="**visible**") button id="@+id/**btn_list_readers**" in "activity_main.xml" layout
**3)** add code below in "MainActivity", public void onClick,**case R.id.btn_list_readers**:

	byte[] readerNames = new byte[1024];
	ret = PCSCNative.SCardListReaders(readerNames);
	showMessage("PCSCNative.SCardListReaders ret : 0x" + Integer.toHexString(ret));
	String[] readers = new String(readerNames).split("\0");
	if (readers.length > 0) showMessage("PCSCNative.SCardListReaders readers[0]: " + readers[0]);

**4)** build and run app, turn on BLE reader
**5)** press **ESTABLISH**, **CONNECT**, select scanned reader, then press **GET READER NAME**. It always returns 0x8010002E (**SCARD_E_NO_READERS_AVAILABLE**).

This is critical for me since after the first successful reading of the certificates from the card in my app it is impossible to do refresh the certificates again. My PKCS11-lib calls **SCardListReaders** again and receives the **NO_READERS_AVAILABLE** and everything stops there (no certificates found). But BLE reader is connected and all ok with it.

boby1975, 2021-12-22:
This problem with **FT_SCardListReaders** was fixed in **FTReaderAPI_1.0.9.4.jar**.
Now I can refresh certificates on the same card but after card changing it's impossible until switch off / on the reader again.
But it's "another story", not so critical and need more deep investigation of behavior of this PCSC realization.
In case of usage Private API (our realization of PCSC) there is no such problems)

FeitianSmartcardReader, 2021-12-24:
@boby1975 thanks for your feedback, will forward to our engineer to handle it, thanks

closed 2022-03-10 by FeitianSmartcardReader