FeitianSmartcardReader/FEITIAN_MOBILE_READERS #33: FT_SCardListReaderGroups: Can not find the bR301 readers

Lectem, 2022-01-20:
My card reader is one with the C18 casing bR301 and the demo application can not connect to it.
It seems that the SDK `FT_SCardListReaderGroups` is looking for devices with names starting by `FT_` or `ID_` but my reader is detected with name `FeiTian73`, which means that it is ignored.

Is there something I need to do for my reader to use the correct name ? Is this a bug with the SDK ?

Note that the Demo code is still using the version 1.0.8 of the jar, and if I try with the 1.0.9 jar, it crashes. (there is no `libFTReaderPCSC_1.0.9.a` nor `libFTReaderPCSC_1.0.9.so`).

FeitianSmartcardReader, 2022-01-20:
[image]
Would you please have a look your firmware version? thanks

Lectem, 2022-01-20:
Somehow I was unable to get the device recognized though USB at all (I tried several times, windows update doesn't give me any new driver, and I tried with multiple USB cables that are known to work with other devices).

I can however connect through the bluetooth driver, but the `FT_bR301_Update_Tool.exe` tool will not display the device firmware version. If I click connect, I get the following error:
[image]

Is there another way to know the device firmware version ?

FeitianSmartcardReader, 2022-01-20:
[image]
Insert reader to your PC, and press power ON button, check the device manager, and after find the hardware ID, share to us, thanks

Lectem, 2022-01-20:
As mentioned, I couldn't get the reader working though USB, I tried with 2 different readers (same model) and 3 different USB cables (all known to work with multiple devices) but windows never detected them.

FeitianSmartcardReader, 2022-01-20:
OK please send mail to me hongbin@ftsafe.com, and let us know where you buy from, sales will support you on this, thanks

Lectem, 2022-01-21:
Closing this in the meantime.

closed 2022-01-21 by Lectem