FeitianSmartcardReader/FEITIAN_MOBILE_READERS #34: mFtReader.power(0) creates : java.lang.NoSuchMethodError: no non-static method "Lcom/ftsafe/comm/CommBase;.getDevType()I

YoeriJPL, 2022-03-09:
When trying to upgrade from 1.0.8 to 1.0.9 (private) we are getting the following error when using the method mFtReader.power(0):
java.lang.NoSuchMethodError: no non-static method "Lcom/ftsafe/comm/CommBase;.getDevType()I

There's a big chance this is due obfuscation or the likes, the code is used in a Xamarin Android project as a binding library

Any idea why this happens + where to find FTReaderAPI_1.0.9.5.jar or higher as a download?

FeitianSmartcardReader, 2022-03-10:
[CardReader_SDK_For_Android_1.0.9.4_20211221.zip](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/files/8219836/CardReader_SDK_For_Android_1.0.9.4_20211221.zip)
@YoeriJPL here is our new lib, if you are using the Private API only, then no need switch to 1.0.9.
1.0.9 fixed bugs with PC/SC API, only very few companies using it. any questions, contact me anytime, thanks

YoeriJPL, 2022-03-10:
Thank you for your quick response

 

Sincerely,

 

Met vriendelijke groeten

Bien à vous

Best regards

 

Yoeri Bernard
Dev team Care4Nurse

	

	Kromstraat 64C 1A
2520 Ranst 
Belgium
W : +32 476 476 343

***@***.***
BE 0862.964.458

YoeriJPL, 2022-03-10:
Thank you for your quick response

Sincerely,

Yoeri Bernard

closed 2022-03-10 by YoeriJPL