FeitianSmartcardReader/FEITIAN_MOBILE_READERS #36: getting PendingIntent Flag error in Android 12 (31 or above)

otashjumaev, 2022-04-20:
this error occuring when using ft.readerFind() in latest Android versions

`W/System.err: java.lang.Exception: java.lang.IllegalArgumentException: uz.newtech.mobilkassa: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
W/System.err: Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
W/System.err:     at com.ftsafe.comm.CommBase.readerFind(CommBase.java:205)
W/System.err:     at com.ftsafe.readerScheme.FTReader.readerFind(FTReader.java:30)
W/System.err:     at uz.newtech.mobilkassa.utils.sam.SamApdu.find(SamApdu.kt:35)
W/System.err:     at uz.newtech.mobilkassa.utils.sam.NewSamHandler.findUsbMode(NewSamHandler.kt:99)
W/System.err:     at uz.newtech.mobilkassa.utils.sam.NewSamHandler$mHandler$1.handleMessage(NewSamHandler.kt:72)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:106)
W/System.err:     at android.os.Looper.loopOnce(Looper.java:226)
W/System.err:     at android.os.Looper.loop(Looper.java:313)
W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:8663)
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
W/System.err: Caused by: java.lang.IllegalArgumentException: uz.newtech.mobilkassa: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
W/System.err: Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
W/System.err:     at android.app.PendingIntent.checkFlags(PendingIntent.java:382)
W/System.err:     at android.app.PendingIntent.getBroadcastAsUser(PendingIntent.java:673)
W/System.err:     at android.app.PendingIntent.getBroadcast(PendingIntent.java:660)
W/System.err:     at com.ftsafe.comm.usb.USB.ft_find(USB.java:140)
W/System.err:     at com.ftsafe.comm.CommBase.readerFind(CommBase.java:180)
`

FeitianSmartcardReader, 2022-05-19:
@otashjumaev hello sir, please share your android version, thanks
And BTW, please try our demo app:
[pcscdemo-1.0.9.6.zip](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/files/8728118/pcscdemo-1.0.9.6.zip)

FeitianSmartcardReader, 2022-05-19:
any question, you can send mail to hongbin@ftsafe.com, thanks

otashjumaev, 2022-05-20:
I am using Android 12, Samsung Galaxy Note 10.
this error only happens if i make targetSdkVersion 32 (does not happen if it is 30 or lower)

FeitianSmartcardReader, 2022-05-20:
our engineer will verify it on next Monday, we will keep you update, thanks

FeitianSmartcardReader, 2022-12-07:
[image]
In android 12, when you do init PendingIntent, need to set FLAG_MUTABLE  and FLAG_IMMUTABLE

this already modified in our latest iReader demo, the new demo will release in this month.

closed 2022-12-07 by FeitianSmartcardReader