FeitianSmartcardReader/FEITIAN_MOBILE_READERS #39: Visual Studio Xamarin Forms example request

Anioz, 2022-08-08:
Is it possible to put an example online for this SDK where Visual studio xamarin forms is used?

A good % of our 3000+ users continue having issue's connecting and we are trying to find out where we are doing something wrong

FeitianSmartcardReader, 2022-08-10:
@Anioz we don't have exist demo code use Xamarin, so cannot help you on this.
And I suggest you can go with latest library to have a test.

we do fixed few bugs which relates the card slot event, you can update it and have a try.

On other hand, I will tell our team to have Xamarin demo code for you too, but it will takes time, once done, I will upload to GITHUB, thanks

Anioz, 2022-08-10:
That would be great, thanks

FeitianSmartcardReader, 2022-08-11:
@Anioz which platform are you using? iOS or Android?

Anioz, 2022-08-11:
The UI code is made with Xamarin Forms (cross platform code) which hooks into the cardreading library's that are used in both sublevels Xamarin.Android & Xamarin.iOS (OS specific code)

FeitianSmartcardReader, 2022-08-11:
noted with thanks,

closed 2022-12-07 by FeitianSmartcardReader