FeitianSmartcardReader/FEITIAN_MOBILE_READERS #40: Does iOS SDK support frameworks?

jukka123456, 2022-12-01:
Hi.

Is it possible to use Feitian SDK in swift framework? I'm able to use Feitian SDK successfully in iOS application but when I include the SDK into framework and try to build it, I get multiple errors from the "libiRockey301_3.5.64_Release.a" file.

`Undefined symbols for architecture arm64:
  "_OBJC_CLASS_$_UIApplication", referenced from:
      objc-class-ref in libiRockey301_3.5.64_Release.a(EADSessionController.o)
  "_UIApplicationDidBecomeActiveNotification", referenced from:
      -[EADSessionController RegisterAccessoryConnectNotification] in libiRockey301_3.5.64_Release.a(EADSessionController.o)
  "_UIApplicationDidEnterBackgroundNotification", referenced from:
      -[EADSessionController RegisterAccessoryConnectNotification] in libiRockey301_3.5.64_Release.a(EADSessionController.o)
  "_UIBackgroundTaskInvalid", referenced from:
      -[EADSessionController HandleAppDidEnterBackground:] in libiRockey301_3.5.64_Release.a(EADSessionController.o)
      ___52-[EADSessionController HandleAppDidEnterBackground:]_block_invoke in libiRockey301_3.5.64_Release.a(EADSessionController.o)
      ___52-[EADSessionController HandleAppDidEnterBackground:]_block_invoke.212 in libiRockey301_3.5.64_Release.a(EADSessionController.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

I also tried to build framework from command line using "xcodebuild -create-xcframework -library libiRockey301_3.5.64_Release.a -headers include -output FeitianLibrary.xcframework" without success. Any help would be appreciated.

FeitianSmartcardReader, 2022-12-05:
@jukka123456 our team will take a look, will get back to you in this week.
in fact, we have demo code in Swift, did you try it? the swift demo in this SDK too.

jukka123456, 2022-12-07:
Yes I did take a look at the swift demo. However it only provides example how to use the SDK in "Application" Xcode project, not in "Framework" project as needed in my use case. Swift example also missed the SDK files and you had to add those manually to get it into working condition. 

Example "Framework" project that runs successfully on latest Xcode and ARM-based Mac would be helpful if possible.

jukka123456, 2023-01-02:
Has there been any progress regarding to this issue?

FeitianSmartcardReader, 2023-01-04:
@jukka123456 currently, it still library on our side, and our engineer will develop such framework and provide demo code, thanks

FeitianSmartcardReader, 2023-01-04:
I will how long it gonna cost in this week?

FeitianSmartcardReader, 2023-01-05:
@jukka123456 the framework will be finish mid of Jan, we will keep you update, thanks

Abdelilahskali, 2023-06-12:
when i run the project iReader : 
```
2023-06-12 16:25:18.008386+0100 iReader_swift[1284:158475] *** Assertion failure in -[CBUUID initWithString:safe:], CBUUID.m:195
2023-06-12 16:25:18.008600+0100 iReader_swift[1284:158475] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'String  does not represent a valid UUID'
*** First throw call stack:
(0x1c6084cb4 0x1bf1243d0 0x1c081a56c 0x1e70f3e24 0x1e70b6fb4 0x1e70b68dc 0x1e70c12e4 0x104c0e0a0 0x104c091ac 0x104befe38 0x104befed0 0x1c8468088 0x1c810490c 0x1c83c6958 0x1c828e0d4 0x1c828d584 0x1c828cbd4 0x1c828cb38 0x1c80eebe0 0x1c75a9074 0x1c75bc5f0 0x1c75cda1c 0x1c75fcff4 0x1c8576e90 0x1c60be6e0 0x1c6125210 0x1c60f50e8 0x1c60fa3ec 0x2015bf35c 0x1c84876e8 0x1c848734c 0x104bf52a8 0x1e55fadec)
libc++abi: terminating due to uncaught exception of type NSException
```

FeitianSmartcardReader, 2023-06-25:
@jukka123456  Hi, it's been a while. Do you solve this issue?

closed 2024-11-06 by FeitianSmartcardReader