FeitianSmartcardReader/FEITIAN_MOBILE_READERS #41: com.ftsafe.readerScheme.FTException: err : 612

NexusNull, 2023-06-02:
I'm having a Problem with my reader. I have already tried a different reader and a different card so I really don't know what the issue is. It used to work before. I'm using the bluetoothLe variant of the reader.
Could you enlighten me what err 612 means?

```
W/System.err: com.ftsafe.readerScheme.FTException: err : 612
W/System.err:     at com.ftsafe.comm.CommBase.readerXfrBlock(CommBase.java:401)
W/System.err:     at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:128)
W/System.err:     at com.example.cardreader.dataReader.ConnectThread.sendXfr(ConnectThread.java:71)
W/System.err:     at com.example.cardreader.dataReader.ConnectThread.queryData(ConnectThread.java:142)
W/System.err:     at com.example.cardreader.dataReader.ConnectThread$1.handleMessage(ConnectThread.java:124)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:106)
W/System.err:     at android.os.Looper.loopOnce(Looper.java:226)
W/System.err:     at android.os.Looper.loop(Looper.java:313)
W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:8757)
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
W/System.err: com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][err : 612][com.ftsafe.readerScheme.FTException: err : 612]
W/System.err:     at com.ftsafe.readerScheme.FTReader.throwFTError(FTReader.java:580)
W/System.err:     at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:135)
W/System.err:     at com.example.cardreader.dataReader.ConnectThread.sendXfr(ConnectThread.java:71)
W/System.err:     at com.example.cardreader.dataReader.ConnectThread.queryData(ConnectThread.java:142)
W/System.err:     at com.example.cardreader.dataReader.ConnectThread$1.handleMessage(ConnectThread.java:124)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:106)
W/System.err:     at android.os.Looper.loopOnce(Looper.java:226)
W/System.err:     at android.os.Looper.loop(Looper.java:313)
W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:8757)
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
```

Stunc0, 2023-06-02:
Hi, 

Is this error comming randomly ?
On which platform you try to read it ?

Stunc0, 2023-06-05:
is the poweron working before ?

FeitianSmartcardReader, 2023-06-05:
the 612 means the communication error, do you have your card ATR? does it happen always or sometimes?

NexusNull, 2023-06-05:
The power on is successful, the ATR is: 3BD097FF81B1FE451F072B

FeitianSmartcardReader, 2023-06-05:
so means the card can be powered on, and when do communication with smart card, it reports 612 error, am I right? what's the lib version?

NexusNull, 2023-06-05:
> The issue is coming no matter the card you put in ?

yes, well I'm only using Health cards

NexusNull, 2023-06-05:
> so means the card can be powered on, and when do communication with smart card, it reports 612 error, am I right? what's the lib version?

yes, I can provide with the commands that I send to the card.
00A4040C06D27600000102
00b08c0000
00b08100000000
00b08200000000

FeitianSmartcardReader, 2023-06-05:
https://play.google.com/store/apps/details?id=com.ftsafe.ireader  this is our demo app, how about test with this App and get the log for us?

NexusNull, 2023-06-05:
[image]

Stunc0, 2023-06-05:
i think we got the same problem, it happens to us on some readers randomly

NexusNull, 2023-06-05:
> that's strange, what's your reader version? can please take photo of your reader and show us the firmware version too? and we also need bluetooth UUID of your reader.
[image]

the app reports the Firmware Version to be 1.14

NexusNull, 2023-06-05:
The uid is ffffffffffffffff

NexusNull, 2023-06-05:
I think this reader is fucked

closed 2023-06-05 by NexusNull