FeitianSmartcardReader/FEITIAN_MOBILE_READERS #42: Use Feitian library on mac os platform

Abdelilahskali, 2023-06-12:
Hello Team,
Is there a way to use the library "libiRockey301_3.5.64_Release.a" on mac os platform, it's generated for iOS platform, can you please generate a mac os one ?

```
In /Users/lexpersona/Documents/AppName/shared/resources/libs/feitian/libiRockey301_3.5.64_Release.a(DUKPT.o), building for macOS, but linking in object file built for iOS Simulator, file '/Users/lexpersona/Documents/AppName/shared/resources/libs/feitian/libiRockey301_3.5.64_Release.a' for architecture x86_64
```
Best Regards

FeitianSmartcardReader, 2023-06-13:
Hi @Abdelilahskali, so far on Apple platform, we only provide iOS SDK for our iOS Reader products.

FeitianSmartcardReader, 2023-06-13:
Hi @Abdelilahskali , for macOS platform, you can check this repo: https://github.com/FeitianSmartcardReader/R301/tree/master/Demo/macOS, the code is written in C.

FeitianSmartcardReader, 2023-06-13:
Hi @Abdelilahskali , our team is working on this issue, please check the update later.

FeitianSmartcardReader, 2023-06-14:
Hi @Abdelilahskali , sorry about linker failed when built for macOS, would you please try the direction in following pic to check if it works?

[image]

For willing to contribute to our project, we very appreciate that.  About the code of repo https://github.com/FeitianSmartcardReader/bR301_iOS/tree/master/libs/source, please wait for our reply later today.

FeitianSmartcardReader, 2023-06-14:
@Abdelilahskali , about Bluetooth, we do have demo for it, check this archive.
[Readerswiftdemo_202306.zip](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/files/11745558/Readerswiftdemo_202306.zip)

closed 2024-11-06 by FeitianSmartcardReader