FeitianSmartcardReader/FEITIAN_MOBILE_READERS #43: Need updates and documentation on Android 13 (SDK 33)

Stunc0, 2023-06-13:
Hi, 

I would like to integrate the sdk in my reactnative app, i tried to take your demo app but it seems that it's not working on SDK 33. No readers founded when i scan... 
Can you update the demo app to see how to scan on theses devices ?  Is it possible to have a documentation ont the permissions we have to ask to scan and connect ? 

Best regards.

FeitianSmartcardReader, 2023-06-14:
Hi @Stunc0 , sorry about this. We're working on this. I'll update the latest Android SDK later today.

Stunc0, 2023-06-14:
Ok thanks to you ;)

FeitianSmartcardReader, 2023-06-14:
Hi @Stunc0 , we updated our Android SDK for new version of API libs (2.0.1.3) and its demo codes, and the iReader apk, in path: https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/Android_SDK/iReader

Please check that.
For the documentation, we'll check to update it soon.

Stunc0, 2023-06-26:
It looks like all have changed in functions signatures/structures...  It's very hard to work in theses condictions, no documentations, no explanations on what have changed. A demo app is not a documentations and it's uite difficult to see whats have changed with the previous SDK.

FeitianSmartcardReader, 2024-11-06:
@Stunc0 sorry for that, suggest you can go with latest SDK.
and developer guide can get from pcscreader.com

closed 2024-11-06 by FeitianSmartcardReader