FeitianSmartcardReader/FEITIAN_MOBILE_READERS #44: FTException: err : 612 with some of the cards

JasKuus, 2023-06-21:
Hi,

I get an exception 
com.ftsafe.readerScheme.FTException: [ERROR] 
[com.ftsafe.readerScheme.FTReader:readerPowerOn][err : 612]
[com.ftsafe.readerScheme.FTException: err : 612]
when I try to power on a card. 

I am using your library **FTReaderAPI_1.0.9.6.jar** in my Android application.

This problem occurs with **Feitian bR301 reader** (151015AA00065H) both with USB and Bluetooth modes.

Power on fails only when I am using our own cards. Cards made by another organization works fine with this reader. I have no problems with our own test cards when I am using readers from other manufacturers (for example ACS). So the Feitian reader should be fine as it works with the other cards. And our own cards should be fine as they work with other readers. Maybe your library is not parsing the ATR from our own cards correctly?

Own card ATR (This does not work):
**3B 9F 95 81 31 FE 9F 00 66 46 53 05 10 00 FF 71 DF 00 00 00 00 00 EC**
JavaCOS A22 dual interface Java card - 150K (JavaCard)
(ATR got with winscard SCardStatus function: 3B 9F 95 81 31 FE 9F 00 66 46 53 05 10 00 FF 71 DF 00 00 00 00 00 EC FF 03 00 00 00 AE B4 D2 01, last bytes are some extra bytes?)

Card made by another organization, ATR (This works):
3B DD 96 00 80 31 FE 45 00 31 B8 64 04 29 EC C1 73 94 01 80 82 48
Finnish Digital and Population Services Agency Organisation Card (eID)
(ATR got with winscard SCardStatus function: 3B DD 96 00 80 31 FE 45 00 31 B8 64 04 29 EC C1 73 94 01 80 82 48 FF FF 03 00 00 00 AE B4 D2 01, last bytes are some extra bytes?)

FeitianSmartcardReader, 2023-06-21:
thanks for your feedback, our engineer is on holiday now, he will back to you next Monday, thanks

FeitianSmartcardReader, 2023-06-25:
@JasKuus 612 indicates it is a IFD_COMMUNICATION_ERROR, would you please check your reader's firmware version. The attached screenshot shows you how to get the firmware version.

[image]

And use this tool to try to power on the card, when the issue occurs, tap the "Log“ on the top left of App screen, and send mail to me: richard@ftsafe.com. 
https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/blob/master/Android_SDK/iReader/BIN/iReader_1.0.1_20230424.apk

JasKuus, 2023-06-26:
Here is the reader's firmware version:

[image]

I have sent you the log via email.

FeitianSmartcardReader, 2023-06-29:
@JasKuus Thank you, I'll let our R&D team to check this issue and back to you asap.

FeitianSmartcardReader, 2023-06-29:
@JasKuus Hi, please use our latest Android SDK lib and demo, the FTReaderAPI updated to 2.0.1.3, in iReader/Lib. 
The link is: https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/Android_SDK/iReader.
If the problem still persists, please let us know.

JasKuus, 2026-04-24:
Hi,

This issue can be closed. FTReaderAPI 2.0.1.7 appears to work better than earlier versions, and we are now able to read the other card as well. It also seems that my very old Feitian bR301 reader is more prone to this problem than newer readers.