FeitianSmartcardReader/FEITIAN_MOBILE_READERS #47: Uncatchable internal lib crash black micro usb ble301 for combination 2.0.1.3 & USB\VID_096E&PID_0624&REV_0112

Anioz, 2023-10-02:
I have provided some testdata in the hopes of making debugging this issue easier:

cardreader USB-C:
USB\VID_096E&PID_0624&REV_0225
Test : OK (very fast readout)

cardreader USB-C:
USB\VID_096E&PID_0624&REV_0222
Test : OK (fast readout)

cardreader Micro USB:
USB\VID_096E&PID_0624&REV_0112
Test : NOK (uncatchable app crash with 2.0.1.3 | slower readout but ok on older firmware 1.0.9.6)

cardreader Micro USB:
USB\VID_096E&PID_0624&REV_0112
Test : NOK (uncatchable app crash with 2.0.1.3 | slower readout but ok on older firmware 1.0.9.6)

All of the older cardreaders type micro usb with product & fw PID_0624&REV_0112 end up in an app crash while ignoring try catches in xamarin.android & pcl xamarin (seems like an uncatchable internal lib error):

[MonoDroid] Java.Lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
[MonoDroid]   --- End of managed Java.Lang.NullPointerException stack trace ---
[MonoDroid] java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
[MonoDroid] 	at com.ftsafe.comm.bt4.BluetoothLeClass.readPowerState(BluetoothLeClass.java:204)
[MonoDroid] 	at com.ftsafe.comm.bt4.BT4.getPowerState(BT4.java:400)
[MonoDroid] 	at com.ftsafe.comm.CommBase.getPowerState(CommBase.java:108)
[MonoDroid] 	at com.ftsafe.readerScheme.FTReader$1.run(FTReader.java:623)
Thread started: <Thread Pool> #14
[MonoDroid] 	at java.lang.Thread.run(Thread.java:1012)
**Java.Lang.NullPointerException:** 'Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference'

Our customers with an older cardreader get this issue only when using the newest driver FTReaderAPI_2.0.1.3.jar, libFTReaderPCSC_2.0.1.3.so

These tests were done on Android 13. We are also open to try out experimental builds if you want us to testrun them as this would provide our numerous customers with a temporary fix until a new main release is published

Stunc0, 2023-11-06:
I think this can be linked and be the same problem than my issue : https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/45

mjiricek, 2024-03-01:
Did you manage to solve this issue? I have the same issue...

Anioz, 2024-03-08:
@mjiricek  We solved it for our clients by rolling back to 1.0.9.6 for now which works with all models.

mjiricek, 2024-03-08:
Thank you for your reply. This isn't solution for me because I also have a [issue](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/48) with 1.0.9.6

FeitianSmartcardReader, 2024-11-06:
@mjiricek @Anioz hello guys. do you still have issue right now?

Anioz, 2024-11-06:
yes i was able to try with driver 2.0.1.5 thanks to Richard, the output gave this:

for android the driver you send has this error only for the MicroUSB type Feitian reader devices (black rectangular color) & presumably older ones:
[AndroidRuntime] FATAL EXCEPTION: Thread-47
[AndroidRuntime] Process: be.jpl_solutions.C4nMobile.client.android, PID: 11234
[AndroidRuntime] java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
[AndroidRuntime] 	at com.ftsafe.comm.bt4.BluetoothLeClass.readPowerState(BluetoothLeClass.java:206)
[AndroidRuntime] 	at com.ftsafe.comm.bt4.BT4.getPowerState(BT4.java:404)
[AndroidRuntime] 	at com.ftsafe.comm.CommBase.getPowerState(CommBase.java:107)
[AndroidRuntime] 	at com.ftsafe.readerScheme.FTReader$1.run(FTReader.java:623)
[AndroidRuntime] 	at java.lang.Thread.run(Thread.java:1012)
[AndroidRuntime] FATAL EXCEPTION: Thread-46
[AndroidRuntime] Process: be.jpl_solutions.C4nMobile.client.android, PID: 11234
[AndroidRuntime] java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
[AndroidRuntime] 	at com.ftsafe.comm.bt4.BluetoothLeClass.readPowerState(BluetoothLeClass.java:206)
[AndroidRuntime] 	at com.ftsafe.comm.bt4.BT4.getPowerState(BT4.java:404)
[AndroidRuntime] 	at com.ftsafe.comm.CommBase.getPowerState(CommBase.java:107)
[AndroidRuntime] 	at com.ftsafe.readerScheme.FTReader$1.run(FTReader.java:623)
[AndroidRuntime] 	at java.lang.Thread.run(Thread.java:1012)

It crashes the app entirely not just the driver itself, but the new USB-C blue square Feitian and the new black rectangular Feitian model devices work very good with the same test and do not crash at all.

Would be lovely if you could take a look at it and build something in so that atleast the whole app will not crash with older devices and if solving the problem is not a possibility a reliable method to detect incompatibility because of this crash would be preferred then if you would allow such an option so we can tell our customers to buy a new Feitian model in a popup or some other idea to rule out the incompatible models before attempting to connect fe.

FeitianSmartcardReader, 2024-11-06:
yes, of course, let our engineer to have a look, and will get back to you soon, thanks.

mjiricek, 2024-11-12:
@FeitianSmartcardReader I got the same error as @Anioz 
```
java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
                                                                                                    	at com.ftsafe.comm.bt4.BluetoothLeClass.readPowerState(BluetoothLeClass.java:206)
                                                                                                    	at com.ftsafe.comm.bt4.BT4.getPowerState(BT4.java:404)
                                                                                                    	at com.ftsafe.comm.CommBase.getPowerState(CommBase.java:107)
                                                                                                    	at com.ftsafe.readerScheme.FTReader$1.run(FTReader.java:623)
                                                                                                    	at java.lang.Thread.run(Thread.java:1012)
```

FeitianSmartcardReader, 2024-11-13:
@mjiricek @Anioz  thanks for the feedbacks and sorry for the inconvenience. team is investigating this, will provide update once we fixed it.

Anioz, 2025-02-25:
Android version libFTReaderPCSC_2.0.1.6.so fixes this specific issue for our users.