FeitianSmartcardReader/FEITIAN_MOBILE_READERS #48: com.ftsafe.readerScheme.FTException: err : 612 - FTReaderAPI library does not work

mjiricek, 2024-03-04:
Hi,

I'm trying to migrate from BR500_AND_BR301BLE_V4.0.6_RELEASE.jar to FTReaderAPI_2.0.1.3.jar. In the new version (2.0.1.3) I got this error:
```
readerXfr_index ====== :0
onResult=======================128
onResult=======================128
recvHead===============:
BT4_BlueToothRead()_len==============:2
BT4_BlueToothRead()_readDataBuffLen==============:12
onResult=======================128
onResult=======================128
recvHead===============:
BT4_BlueToothRead()_len==============:0
BT4_BlueToothRead()_readDataBuffLen==============:10
com.ftsafe.readerScheme.FTException: err : 612
at com.ftsafe.comm.CommBase.readerXfrBlock(CommBase.java:401)
at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:128)
```
I also got the same error in version 1.0.9.6... What does error 612 mean?
I have these BT readers:

1. [image]
2. [image]

Tested on Huawei Mate 20 Pro with Android 9 and EMUI 9.1.0. 

Thanks in advance for trying to help me solve this problem.

Stunc0, 2024-03-04:
Is this issue happends on each read ? 
Also on 2.25 firmware reader ?

richard605, 2024-03-05:
@mjiricek Sorry for your inconvenience.
The error code 612 could be output in several routines. It needs to be investigated by team.
May I know when migrate project code, did you refer to the iReader demo project code? 
Here is the demo project code: https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/Android_SDK/iReader, please have a reference.
Also, my email address is: richard@ftsafe.com, please feel free to contact with me.

mjiricek, 2024-03-05:
@richard605 Yes, I used the demo project code to migrate the libraries.
@Stunc0 No, before this issue a I send several instructions to the card successfully. The error occurred on both readers (FW 2.25, FW 1.10) when I used library 1.0.9.6. When I used library 2.0.1.3, I got another [error](https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/47) on the reader with FW 1.10

richard605, 2024-03-05:
@mjiricek sorry to hear that. Since it is an old model, and we don't have any in stock, I am trying to manage to find one same reader with bluetooth firmware 1.1.00, but I can't find one by now. Let me talk with team what we can do next.

Stunc0, 2024-03-05:
@mjiricek 

> When I used library 2.0.1.3, I got another https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/issues/47 on the reader with FW 1.10
Yes, readers with FW 1.10 not working with this SDK version... 

@richard605 , Ask to Ben from Feitian, but it's possible we give him some of them (FW 1.10) during our meeting in France.
And we have some of them to exchange as well.

mjiricek, 2024-03-18:
@richard605 Did you manage to reproduce the issue?

mjiricek, 2025-02-28:
With the new version 2.0.1.6 I still got the same issue on both readers @richard605 

```
 W  com.ftsafe.readerScheme.FTException: err : 612
 W  	at com.ftsafe.comm.CommBase.readerXfrBlock(CommBase.java:419)
 W  	at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:135)
```

mjiricek, 2026-09-03:
@richard605 @nmauffre @FeitianSmartcardReader The bug can still be reproduce with **FTReaderAPI 2.0.1.7** (BR301 BLE), and I have now been able to narrow down the root cause. I hope this finally gives the team something concrete to reproduce with.

- #### I’ve noticed that the trigger is the APDU length, not a specific card.
> Error 612 only occurs on the first command APDU that reaches ~254 bytes (CCID message ~264 bytes). Every shorter command in the same session succeeds. Including slow ones (e.g. a 149-byte GENERAL AUTHENTICATE that takes the card >1 s to compute succeeds), so this is not a card-computation timeout. It fails reproducibly once the command crosses the ~254-byte mark. In my case the failing command is a chained, secure-messaging PSO: VERIFY CERTIFICATE (`CLA=1C INS=2A P1=00 P2=BE Lc=F9`, i.e. 249 bytes of data).
This should be reproducible with any ~254-byte-or-longer command APDU sent to a card over BR301 BLE. You should not need my specific card to trigger it.

By logging the BT4 internal messages, I captured the reader's actual CCID reply to the failing command:
```
PC → RDR:  254-byte XfrBlock dispatched
RDR → PC:  80 00000000 00 2A 40 FE 00
```
Decoding this `RDR_to_PC_DataBlock`:
- `dwLength = 0` → no data returned from the card
- `bStatus = 0x40` → command failed, ICC present and active
- `bError  = 0xFE` → **ICC_MUTE** (card did not respond)

So the reader forwards the block and then reports that the card never answered. A follow-up GetSlotStatus also returns `0x40 / 0xFE`; the card stays mute until the connection is closed. This strongly suggests the card received a truncated or malformed block and is waiting for the rest of the data field until its waiting time expires.