FeitianSmartcardReader/FEITIAN_MOBILE_READERS #49: Cannot connect to Feitian bR301 BLE on Android

hoschbel, 2024-03-06:
Dear support,

We bought a brand new Feitian bR301 BLE (so I suppose that the firmware is up-to-date). But I'm not able to connect to that device. It simply cannot find the device. That's what I did:

1. Downloaded and installed the Feitian Test App from Google Play Store: entered "com.ftsafe.cardreader" as search string, found app "CardReader" and installed it.
2. Switched the reader on by pressing the power button (blue light is flashing)
3. Started the app and:
- Selected "BT4" as device type
- pressed button "find"
- scanning process is running endlessly. Nothing is found!

The same is true, if I use another bluetooth test app like "Bluetooth LE Scanner". The feitian scanner is not found! On the other hand, other LE devices are working fine!
When going to the Android Bluetooth settings --> connect new device, I can see the Feitian reader (entry "FT_FC4979...."). When clicking the entry, it says "You need an app to connect to this device". But this is OK, since it's BLE.
I testet it on 2 Android devices: Samsung tablet Galaxy Tab S5e (Android 11) and an Android smartphone with Android 10... both supporting BLE.

Could you please advice.

Best regards, Martin
(We are a software company building a german POS software. We have years of Android knowledge and are familiar with BLE stuff, normally)

hoschbel, 2024-03-06:
On Windows I'm able to detect the device with the Windows app "Bluetooth LE Explorer". I just checked the firmware version: it's the new version, the UUID is 46540001-0002-00c6-0000-465453414645.
Still, the Feitian test app on Android is not able to detect the device. Tested with a third device Samsung XCover 5 (Android 13).

richard605, 2024-03-08:
@hoschbel Hi, sorry for your inconvenience. 
Could you please try this test demo app from Google Play Store: https://play.google.com/store/apps/details?id=com.ftsafe.ireader

richard605, 2024-03-08:
And for the reader's Android SDK, and its demo project code, please check: https://github.com/FeitianSmartcardReader/FEITIAN_MOBILE_READERS/tree/master/Android_SDK/iReader

hoschbel, 2024-03-08:
Thanks for your answer. But unfortunately it still does not work. 

I did the following:
1. Installed "iReader" app from Google Play Store and granted all requested permissions.
2. Switched the reader on. Blue light is blinking. In the Android bluetooth settings, the "FT_...."-device is listed.
3. Started the iReader app.
4. It says "Waiting for connection... No reader detected"....forever.

I tried the same with the iReader-APK from Github. Same result.
Tried with another Android device. Same result...

richard605, 2024-03-11:
Do you have another reader, did they have same phenomenon?
Can you plug this reader into a Windows PC to have a check the reader device **VID/PID/REV** info from DeviceManager. 
[image]

richard605, 2024-03-11:
> On Windows I'm able to detect the device with the Windows app "Bluetooth LE Explorer". I just checked the firmware version: it's the new version, the UUID is 46540001-0002-00c6-0000-465453414645. [...]

Could you please provide the Bluetooth firmware version?

richard605, 2024-03-11:
And, moreover, to check if this reader have Bluetooth connection issue or not, do you have an Apple device(iPhone or iPad), may I know this reader can connect with iPhone/iPad device or not, with our iReader app from [iOS Store](https://apps.apple.com/us/app/smartcard-reader/id525954151)

hoschbel, 2024-03-28:
Hi Richard,
Sorry for my late reply, I've been on vacation.
As you proposed, I checked the USB VID/PID. I have 2 readers: an old reader from 2019 with old firmware version (I don't expect that this works) and a brand-new reader from 2024 with new firmware version.

BOTH readers don't work and show the same VID/PID (see attachments):   
- older reader: USB\VID_096E&PID_0624&REV_0110
- new reader:   USB\VID_096E&PID_0624&REV_0225

These are different PID's than in your screenshot! Maybe this is the problem?

Concerning your question "Could you please provide the Bluetooth firmware version?": what do you mean excatly and how to get this information?

I don't have apple devices. Sorry.

BR, Martin

[image]
[image]

hoschbel, 2024-04-15:
Hi Richard,
It would be kind, if you could check this issue, please. We need a solution here...
BR, Martin

richard605, 2024-04-23:
> Thanks for your answer. But unfortunately it still does not work.
>
> I did the following:
>
> 1. Installed "iReader" app from Google Play Store and granted all requested permissions.
> 2. Switched the [...]

Hi @hoschbel , please make sure the bR301BLE's Bluetooth not automatically connected to any other device, according to experience, if once paired & connected bR301BLE with laptop (with Bluetooth enabled), next time when the bR301BLE starts up, it quickly and automatically connects to laptop again, in this situation, it won't connect to the other device any more. 
So can you try again with iReader demo app from [Google Play Store](https://play.google.com/store/apps/details?id=com.ftsafe.ireader), and please disable any device Bluetooth that the bR301BLE paired/connected before?

closed 2024-11-06 by FeitianSmartcardReader