FeitianSmartcardReader/FEITIAN_MOBILE_READERS #50: FTException: err : 612

StijnWauterss, 2024-06-06:
Hi, I'm using the latest Android SDK (2.0.1.3) and I'm getting the following error when sending APDU command over CardReader_XFRSend() method (see below). The APDU command I'm sending is the following: "00A4000C023F00". This is how I initialise the method: mFtReader.ReaderXfr("DEVICE", slotIndex, send); I've already tried the previous library aswel with the same outcome. Thanks in advance! 

 at Java.Interop.JniEnvironment+InstanceMethods.CallObjectMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00068] in /Users/runner/work/1/s/xamarin-android/external/Java.Interop/src/Java.Interop/obj/Release/JniEnvironment.g.cs:11524 
  at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeVirtualObjectMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x00036] in /Users/runner/work/1/s/xamarin-android/external/Java.Interop/src/Java.Interop/Java.Interop/JniPeerMembers.JniInstanceMethods_Invoke.cs:867 
  at Com.Ftsafe.ReaderScheme.FTReader.ReaderXfr (System.String device, System.Int32 slotIndex, System.Byte[] send) [0x00051] in C:\Users\swauters\source\repos\FtReaderApi\FtReaderApi\obj\Debug\generated\src\Com.Ftsafe.ReaderScheme.FTReader.cs:931 
  at android_arguseid.CardReaderInterfaceImpl.CardReader_XFRSend (System.String sendata) [0x00040] in C:\Users\swauters\source\repos\android_arguseid\android_arguseid\CardReaderInterfaceImpl.cs:507 
  --- End of managed Com.Ftsafe.ReaderScheme.FTException stack trace ---
com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerXfr][err : 612][com.ftsafe.readerScheme.FTException: err : 612]
	at com.ftsafe.readerScheme.FTReader.throwFTError(FTReader.java:663)
	at com.ftsafe.readerScheme.FTReader.readerXfr(FTReader.java:142)
	at crc64ee486da937c010f4.ButtonRenderer.n_onClick(Native Method)
	at crc64ee486da937c010f4.ButtonRenderer.onClick(ButtonRenderer.java:107)
	at android.view.View.performClick(View.java:7465)
	at android.view.View.performClickInternal(View.java:7442)
	at android.view.View.access$3600(View.java:813)
	at android.view.View$PerformClick.run(View.java:28394)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:250)
	at android.app.ActivityThread.main(ActivityThread.java:7806)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)

richard605, 2024-06-17:
Hi @StijnWauterss ,

> I'm using the bR301 and bR301 BLE reader and try to read out a Belgian eID card over Bluetooth. Connecting w/ these two devices works fine.

The "w/" means Windows?

richard605, 2024-06-17:
currently, we don't have newer reader firmware, let's aim to this issue you encountered. I want to know more context about this output error lines, and can we get a logcat record for team that includes such error information and all its continuous commands & response APDUs before. Thanks.

abduqahhoruz, 2025-04-04:
readPowerOn() - Open Device Failed -> com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerPowerOn][err : 612][com.ftsafe.readerScheme.FTException: err : 612]

abduqahhoruz, 2025-04-04:
> readPowerOn() - Open Device Failed -> com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerPowerOn][err : 612][com.ftsafe.readerScheme.FTException: err : 612]

Occured on android when trying to {ftReader?.readerPowerOn(name, 0).log("power_on", "APDU")}