FeitianSmartcardReader/FEITIAN_MOBILE_READERS #51: Broadcasts receiver in Android 14: RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED

otashjumaev, 2024-08-22:
Hi, I am using latest version 2.0.1.3 for Android. On Android 14 devices, `FTReader(context, mHandler, DK.FTREADER_TYPE_USB) `  throwing this exception:

`80=> [LOG]com.ftsafe.readerScheme.FTReader:showErrLog--->ERROR::com.ftsafe.readerScheme.FTReader:<init> [x.x.x. One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts][java.lang.SecurityException: x.x.x. : One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts]`

I think it is from new requirements, and this row is causing this problem:

`            this.tContext.getApplicationContext().registerReceiver(mUsbReceiver, filter);` 
in USB class (rows: 143, 156).

> As discussed at Google I/O 2023, registering receivers with intention using the RECEIVER_EXPORTED / RECEIVER_NOT_EXPORTED flag was introduced as part of Android 13 and is now a requirement for apps running on Android 14 or higher (U+).
> 
> If you do not implement this, the system will throw a security exception.

richard605, 2024-08-23:
Hi  @otashjumaev , we recently updated preview version for adapting Android 14 (API Level 34), please download here and have a test:  https://docs.ftsafe.cn/s/WaAYFA6Bx7kSncX
Looking forward to your feedback, if you encountered any problems with it.

otashjumaev, 2024-08-23:
Hi, thank you for your quick response.

This new version is still throwing the same Exception.

`DK_80: [LOG]com.ftsafe.readerScheme.FTReader:showErrLog--->ERROR::com.ftsafe.readerScheme.FTReader:<init> [xxx: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts][java.lang.SecurityException: xxx: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts]`

Stunc0, 2024-10-08:
Hi @richard605, any update on this ?