FeitianSmartcardReader/FEITIAN_MOBILE_READERS #52: Feitian iR301-UD-C - Reading the Health Card with APDU

Kev2012, 2024-09-06:
Dear Team,

We would like to use your Feitian iR301-UD-C card reader to retrieve data from health cards stored in the public directory. The following information is what we intend to read and display in our own app:

- First and last name
- Date of birth
- Address (country, street, postal code, and city)
- Insurance provider
- Cost carrier identification
- Insurance number
- Status

Unfortunately, we are not very experienced in programming with card readers, so we urgently need assistance in this area. When we attempt to start your reader using the app provided, we receive a message stating that the reader has not been initialized. 

Would it be possible for you to assist us in reading the health card? We are using Android with the latest SDK 34, and we develop our apps in Kotlin and Jetpack Compose.

Best regards,  
Kevin

Kev2012, 2024-09-06:
Hello Ben,

Thank you for your answer. We have installed PCSC-Lite on our Android phone. However, when we click on “Connect” after the reader is connected to the phone, it says:
list readers ret: 0x80100017

It indicates that no reader is available. We believe the issue is that Android SDK 34 requires a permission request that must be accepted on the phone itself.

We need to know how the reader connects to the eHealthcare data. We will receive the APDU commands from the German Gematik in the coming days.

If you have any further questions or need additional adjustments, feel free to ask!

We send you our Project. Hope you can check it and help us. :) Because it is to Big here the Link:
https://we.tl/t-1meYiHFVUv

best Regards
Kevin

FeitianSmartcardReader, 2024-09-09:
Hello Kevin

Was busy today, do you have skype account? We can have a group and discussion with engineer.
Here is my account ***@***.***

Thanks

Kev2012, 2024-09-09:
Hello Ben,

your Mailadress is with ** can you write the name than at for @ than the rest from the Mail?

Thank you.

SelcukOzkara, 2024-09-09:
Hello Ben & Kev2012,

I managed to send a request using APDU commands and receive data from the eGK with the demo code. Now, I’m trying to integrate the code into my existing Jetpack Compose project, but I’m encountering the following error:

com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerFind][Attempt to invoke virtual method 'int com.ftsafe.comm.CommBase.readerFind()' on a null object reference][java.lang.NullPointerException: Attempt to invoke virtual method 'int com.ftsafe.comm.CommBase.readerFind()' on a null object reference]

richard605, 2024-09-09:
hello @Kev2012, let's talk via Skype, Richard (skype name: "live:1yvt0s") and Ben (Skype name: "xwhbin").

also, @SelcukOzkara please add us Skype account too.

That'll be more efficient. Thanks you guys.

closed 2024-11-06 by FeitianSmartcardReader