FeitianSmartcardReader/FEITIAN_MOBILE_READERS #54: Feitian iR301 SmartCard Reader Does not Get Powered On When Connecting to iOS Device

bikashpoudel17, 2024-10-03:
I purchased two Feitian USB-C iR301 smart card readers. When I connect it to my MAC and ran tests using PCSC tool. But, when I connect it to my iPhone (which has lightning port)  using a USB-C to Lightning Dongle, the smart card reader did not get powered on. I tried on my Nokia Phone which has iReader and has USB-C port, I experience similar issue. Am I missing anything?

bikashpoudel17, 2024-10-03:
I was able to connect it to my iPad which has USB-C port and able to interact with the card reader using iReader App.

Stunc0, 2024-10-07:
If i don't tell a mistake, the reader with lithning cannot be compatible with iPhone with usb-c port (even with a dongle).
You must use the usb-c reader from feitian on iphone with usb-c port.

bpoudel7, 2024-10-22:
This issue is resolved thank you.

closed 2024-11-06 by FeitianSmartcardReader