FeitianSmartcardReader/FEITIAN_MOBILE_READERS #56: FTSAFE Error 5002 occurs periodically upon insert of a card and shuts down the device

Anioz, 2025-02-25:
FTSAFE Error 5002 occurs periodically when we insert a card for readout causing a shutdown of the cardreader, its notable that this only is occuring for a specific reader type of our last most recent batches and this error also occurs on iOS which has a complete seperate driver making us personally believe the problems origin might be in the firmware but offcourse no way for us to be sure

The extended output on the console for this occurence:
[CompatibilityChangeReporter] Compat change id reported: 263076149; UID 10205; state: ENABLED
[0:] **** StartScan establishing context
[BluetoothAdapter] isLeEnabled(): ON
[BluetoothLeScanner] could not find callback wrapper
[BluetoothAdapter] isLeEnabled(): ON
[BluetoothLeScanner] onScannerRegistered() - status=0 scannerId=1 mScannerId=0
[FTSAFE] find ble device : FT_F9139BFC7EF2
Thread started:  #40
[0:] **** Bluetooth Connect event
[0:] **** ConnectProcedure started
[0:] **** Favorite cardreader known
[0:] **** BLE Cardreader establishing connection
[BluetoothAdapter] isLeEnabled(): ON
[CompatibilityChangeReporter] Compat change id reported: 265103382; UID 10205; state: ENABLED
[BluetoothGatt] connect() - device: XX:XX:XX:XX:7E:F2, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=9316f15f-3d5b-45cd-a685-58fcd76c928d
[BluetoothGatt] onClientRegistered() - status=0 clientIf=6
[0:] **** Bluetooth Connect event
[0:] **** Bluetooth Connect event
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=6 device=F9:13:9B:FC:7E:F2
[FTSAFE] [BLE-onConnectionStateChange]status : 0, newState : 2
[BluetoothGatt] discoverServices() - device: XX:XX:XX:XX:7E:F2
[0:] **** Intent Action ACL Connect to FT_F9:13:9B:FC:7E:F2
[0:] **** Intent Action ACL Connect - TryOpen
[BluetoothGatt] onConnectionUpdated() - Device=F9:13:9B:FC:7E:F2 interval=6 latency=0 timeout=500 status=0
[0:] **** ConnectProcedure Stopped because of connection found
[BluetoothGatt] onSearchComplete() = Device=F9:13:9B:FC:7E:F2 Status=0
[FTSAFE] [BLE-onServicesDiscovered]status : 0
[BluetoothGatt] setCharacteristicNotification() - uuid: 46540003-0002-00c4-0000-465453414645 enable: true
[BluetoothGatt] setCharacteristicNotification() - uuid: 46540003-0002-00c4-0000-465453414645 enable: true
[BluetoothGatt] onConnectionUpdated() - Device=F9:13:9B:FC:7E:F2 interval=39 latency=0 timeout=500 status=0
[FTSAFE] [BLE-onCharacteristicChanged]5002
[FTSAFE] NEW:5002
[FTSAFE] [BLE - onCharacteristicRead] status : 0
[.client.android] Explicit concurrent copying GC freed 1484KB AllocSpace bytes, 9(180KB) LOS objects, 75% free, 8421KB/32MB, paused 35us,37us total 96.003ms
[BluetoothGatt] onConnectionUpdated() - Device=F9:13:9B:FC:7E:F2 interval=27 latency=0 timeout=400 status=0
[BluetoothGatt] onClientConnectionState() - status=8 clientIf=6 device=F9:13:9B:FC:7E:F2
[FTSAFE] [BLE-onConnectionStateChange]status : 8, newState : 0
[BluetoothGatt] close()
[BluetoothGatt] unregisterApp() - mClientIf=6
[0:] **** Intent Action ACL Disconnect from FT_F9:13:9B:FC:7E:F2

the output of the android device log:
Time	Device Name	Type	PID	Tag	Message
02-25 10:27:14.880	Samsung SM-N970F	Error	22158	FTSAFE	[BLE - onCharacteristicRead] status : 0
02-25 10:27:14.813	Samsung SM-N970F	Error	22158	FTSAFE	NEW:5002
02-25 10:27:14.813	Samsung SM-N970F	Error	22158	FTSAFE	[BLE-onCharacteristicChanged]5002
02-25 10:27:11.099	Samsung SM-N970F	Error	22158	FTSAFE	[BLE - onCharacteristicRead] status : 0
02-25 10:27:11.036	Samsung SM-N970F	Error	22158	FTSafe	BT4_RecvData:81000000000065010000
02-25 10:27:11.035	Samsung SM-N970F	Error	22158	FTSAFE	NEW:81000000000065010000
02-25 10:27:11.034	Samsung SM-N970F	Error	22158	FTSAFE	[BLE-onCharacteristicChanged]81000000000065010000
02-25 10:27:11.033	Samsung SM-N970F	Error	22158	FTSAFE	[BLE - onCharacteristicRead] status : 0
02-25 10:27:10.981	Samsung SM-N970F	Error	22158	FTSAFE	[BLE-onCharacteristicWrite]63000000000065000000
02-25 10:27:10.980	Samsung SM-N970F	Error	22158	FTSAFE	timeout:3000
02-25 10:27:10.978	Samsung SM-N970F	Error	22158	FTSAFE	[BT4-SEND]63000000000065000000

When hooked up to the Windows computer this is the hardware/ven id with rev:
USB\VID_096E&PID_0624&REV_0225 (Reseller Zetes)

Can you offer us more insight in what is happening here and on how to possibly solve it?

FeitianSmartcardReader, 2025-02-26:
Hi @Anioz, sorry for such experience.  thank you for the feedback. I've forward the logs also the phenomenon  situation corresponding to the operations to team, we will carry out the analysis and then back to you ASAP.

Anioz, 2025-02-28:
might be helpfull in troubleshooting the above issue is to expose your test device to a lower sdk: i tested the driver on an Android 10 and a new error FTSAFE error 5003 showed itself along with the FTSAFE error 5002 which was already reported above. 
Symptom is that it does connect but in this case the insert event wont trigger, ive verified this in the IDE:
[0:] **** Bluetooth Connect event
[BluetoothGatt] onConnectionUpdated() - Device=DC:81:0D:BE:B9:0D interval=27 latency=0 timeout=400 status=0
[FTSAFE] [BLE-onCharacteristicChanged]5003
[FTSAFE] NEW:5003