FeitianSmartcardReader/FEITIAN_MOBILE_READERS #57: FtReader?.readerFind() Targeting U+ (version 34 and above)

abduqahhoruz, 2025-04-04:
findUsbMode() - Exception: com.ftsafe.readerScheme.FTException: [ERROR][com.ftsafe.readerScheme.FTReader:readerFind][java.lang.IllegalArgumentException: uz.newtech.mobilkassa: Targeting U+ (version 34 and above) disallows creating or retrieving a PendingIntent with FLAG_MUTABLE, an implicit Intent within and without FLAG_NO_CREATE and FLAG_ALLOW_UNSAFE_IMPLICIT_INTENT for security reasons. To retrieve an already existing PendingIntent, use FLAG_NO_CREATE, however, to create a new PendingIntent with an implicit Intent use FLAG_IMMUTABLE.][java.lang.Exception: java.lang.IllegalArgumentException: uz.newtech.mobilkassa: Targeting U+ (version 34 and above) disallows creating or retrieving a PendingIntent with FLAG_MUTABLE, an implicit Intent within and without FLAG_NO_CREATE and FLAG_ALLOW_UNSAFE_IMPLICIT_INTENT for security reasons. To retrieve an already existing PendingIntent, use FLAG_NO_CREATE, however, to create a new PendingIntent with an implicit Intent use FLAG_IMMUTABLE.] (Ask Gemini)
                                                    	at com.ftsafe.readerScheme.FTReader.throwFTError(FTReader.java:663)
                                                    	at com.ftsafe.readerScheme.FTReader.readerFind(FTReader.java:36)

1yvT0s, 2025-04-04:
Hi @abduqahhoruz , which SDK version did you use in your project?

abduqahhoruz, 2025-04-04:
> Hi [@abduqahhoruz](https://github.com/abduqahhoruz) , which SDK version did you use in your project?

Hi. I used 2.0.1.6 version

1yvT0s, 2025-04-04:
Thanks, I'll let team check this next Monday.

abduqahhoruz, 2025-04-08:
Hi @1yvT0s . Any updates on this issue?

FeitianSmartcardReader, 2025-04-09:
Hi @abduqahhoruz Team is working on this. Will provide the SDK update once we got it later.

FeitianSmartcardReader, 2025-04-14:
Hi @abduqahhoruz please use this update for a test: https://docs.ftsafe.cn/s/Yyzm8RKHkxcg4if
Let me know if this issue still or not. Thanks.