FeitianSmartcardReader/FEITIAN_MOBILE_READERS #58: Support 16 KB page sizes

abduqahhoruz, 2025-06-20:
As you know, starting Android 15 -  16 Kb page support is mandatory. https://android-developers.googleblog.com/2025/05/prepare-play-apps-for-devices-with-16kb-page-size.html.   So please provide us with the library that supports 16Kb page size ASAP.

FeitianSmartcardReader, 2025-06-20:
@abduqahhoruz thanks for your feedback, will let team do the check.

FeitianSmartcardReader, 2025-07-03:
@Peanutbutteer Hi the preview update version will be provided next week. Thx

abduqahhoruz, 2025-07-22:
@FeitianSmartcardReader Hi is there any updates on 16kb support?

abduqahhoruz, 2025-09-30:
Here is the NDK version I have been using: 

[image] @FeitianSmartcardReader

sbichara, 2025-10-30:
Hi Richard, may I ask you to remove my phone number and company name from the last conversation?
Thank you

richard605, 2025-10-31:
Hi @sbichara , I don't see any remove or edit operation item from the up-right operation drop down menu of your that comment. Since I already recorded this issue you fed back into our bug system, I think you may remove that comment if needed.

richard605, 2025-11-04:
@sbichara , for this issue, any logs from logcat of AndroidStudio when the issue occurs?

> thanks [@sbichara](https://github.com/sbichara), let me forward to team, then we can begin the code review with "Happens at the moment the user want to establish a connection to the reader". I will update this thread for any progress.